riscv-non-isa/riscv-iommu #256: Some questions about the record generate in the fault queue

wangyongzhen0322, 2023-09-06:
> The IOMMU may identify multiple requests as having detected an identical fault. In such cases the
IOMMU may report each of those faults individually, or report the fault for a subset, including one,
of requests.

I don't quite understand the above sentence. Are those requests errors detected at the same time？
What is "report the fault for a subset"?Does it mean that only one report is generated for those requests?
And if multiple faults detected at the same time, should they all generate their own record  to the FQ? And what order do they follow?

ved-rivos, 2023-09-06:
Imagine an IOMMU overseeing a PCIe root port, to which a device, like an NVMe controller, is connected. This device can initiate a series of writes. If tasked with copying 4 KiB of data from storage to memory with a maximum payload of 512 bytes, it would generate 8 memory writes.

Should the IOMMU encounter an address translation fault, and since all these writes target the same page, they would all identify the same fault. Depending on the IOMMU, it might report:

1. All 8 faults individually.
2. A single consolidated fault.
3. Some fraction of the 8 identical faults.

Some IOMMUs can only handle one request at a time, meaning they'd report every fault. However, more advanced IOMMUs can process multiple requests concurrently, potentially consolidating these faults.

I hope this clarification helps.

wangyongzhen0322, 2023-09-06:
I see, thanks a lot.

ved-rivos, 2023-09-06:
Thanks for raising this question, @wangyongzhen0322. Glad that was helpful. If you have any more questions or concerns in the future, please don't hesitate to ask. Closing this issue now

closed 2023-09-06 by ved-rivos