riscv-non-isa/riscv-iommu #259: Some questions about IOMMU validation commands

18772820305, 2023-09-21:
1. The software has configured the ATS.INVAL command to CQ, and IOMMU fetched and decodes the ATS.INVAL command. It will issue an ATS invalidation request message to RC. Does IOMMU need to invalidate the local cache content of IOMMU at this time.

2. After configuring the IOINVAL.VMA command, does the software need to ensure that the DMA address translation before the IOINVAL.VMA command is completed to avoid excessive invalid TLB causing TLB miss before the IOINVAL.VMA command.

3. Does IOMMU need to notify the software after completing the IOINVAL.VMA command so that the software can configure a new DMA.  To avoid using outdated TLB content in DMA address translation after the IOINVAL.VMA command.

ved-rivos, 2023-09-21:
> The software has configured the ATS.INVAL command to CQ, and IOMMU fetched and decodes the ATS.INVAL command. It will issue an ATS invalidation request message to RC. Does IOMMU need to invalidate the local cache content of IOMMU at this time.

The `ATS.INVAL` command instructs the IOMMU to send an "Invalidation Request" message to the device ATC identified by the combination of DSEG (if valid) and RID. It does  not cause any invalidation in the IOMMU address translation caches.

> Does IOMMU need to notify the software after completing the IOINVAL.VMA command so that the software can configure a new DMA. To avoid using outdated TLB content in DMA address translation after the IOINVAL.VMA command.

If software needs to know when the command has completed, it should queue a `IOFENCE.C` command. Completion of an `IOFENCE.C` command implies that all previous commands fetched from the CQ have been completed and committed. The completion of `IOFENCE.C` can be observed by `cqh` advancing past the index where it was placed in CQ or by requesting an explicit notification by setting `AV=1` to request a write of the data specified in the command to the address specified in the command. Please also see section 6.3.6 about the types of changes allowed to an PTE without first clearing the `V` bit in the PTE.

> After configuring the IOINVAL.VMA command, does the software need to ensure that the DMA address translation before the IOINVAL.VMA command is completed to avoid excessive invalid TLB causing TLB miss before the IOINVAL.VMA command.

This question is not clear to me. Please elaborate further.

18772820305, 2023-09-22:
Thank you for your reply. After the software configures an IOINVAL.VMA command in CQ, is it necessary to configure a new IOFENCE command before configuring the new DMA transfer to ensure that the IOᛁNVAL.VMA command is submitted?

ved-rivos, 2023-09-22:
> After the software configures an IOINVAL.VMA command in CQ, is it necessary to configure a new IOFENCE command before configuring the new DMA transfer to ensure that the IOᛁNVAL.VMA command is submitted?

Typically, `IOTINVAL.VMA` serves to invalidate the IOMMU's address translation cache after modifying the pertinent Page Table Entry (PTE), ensuring the IOMMU operations are synchronized with the change. If a subsequent DMA operation must take into account this updated PTE, it's crucial to confirm that `IOTINVAL.VMA` has fully executed. In such a scenario, employing `IOFENCE.C` becomes requisite to guarantee the completion of these operations.

ved-rivos, 2023-09-26:
Thanks for raising this question, @18772820305. Hope that was helpful. If you have any more questions or concerns in the future, please don't hesitate to ask. Closing this issue now

closed 2023-09-26 by ved-rivos