riscv-non-isa/riscv-iommu #260: Why are the ITag fields and related errors that appear on ATS invalidation completion not mentioned in the IOMMU?

wangyongzhen0322, 2023-09-22:
I've been looking at the PCIE ATS Specification. The specification has a field called ITag for the invalidation completion message. This field is described as follows.

> The ITag field is constrained to the values 0 to 31 and is used by the TA to uniquely identify requests it issues. A TA must
> ensure that once an ITag is used, it is not reused until either released by the corresponding Invalidate Completions or by
> a vendor-specific timeout mechanism (see below).

There is also an error possible with this field. The error is described below.

> A TA that receives an Invalidation Completion for an ITag that has no outstanding Invalidation Request shall report this
> error using implementation specific mechanisms.

But I did not find the relevant description in the entire IOMMU document. Did you forget it or deliberately omit it? If it is the latter, what are the considerations?

ved-rivos, 2023-09-22:
The objective isn't to duplicate the content found in PCIe specifications, which should be obtained directly from the PCIe SIG for authoritative information. As for the error query, the subsequent sentence following the one you cited sheds light on the possible mechanisms under discussion. This spec provides adequate guidance to produce an IOMMU compliant with the standard.

wangyongzhen0322, 2023-09-22:
Thank you for your answer. I'm sorry that I didn't understand the purpose of the spec correctly.

closed 2023-09-22 by wangyongzhen0322

ved-rivos, 2023-09-22:
No worries about the misunderstanding; specs can often be dense and open to interpretation. Glad my explanation clarified things for you. If you have more questions or need further elaboration, feel free to ask.