riscv-non-isa/riscv-iommu #262: Why the status is set to Success,when no other faults were encountered but the "Page Request" could not be queued due to the page-request queue being full (pqt == pqh - 1) or had a overflow (pqcsr.pqof == 1)?

wangyongzhen0322, 2023-10-26:
As explained below, when an overflow occurs, the page request message is discarded, that is, the PRG is not fully received by the software. In my opinion, the status should also be set to Response failure at this time. I cannot understand why status is set to Success.
> While either error bit is set in pqcsr, the IOMMU discards all subsequent "Page Request" messages, including the message that caused the error bits to be set.

ved-rivos, 2023-10-26:
Good question. This behavior is specified to allow the OS to be able to recover from a temporary overflow of the Page Queue.

Lets first recap the page fault protocol. The device makes a ATS translation request and receives insufficient permissions in response. This leads the device to send a Page Request message to the operating system asking for the desired permissions. 

The Page Request may be responded three ways a) Success b) Invalid Request c) Response failure. The Invalid Request response is for cases where request cannot be granted and the ATS request will never succeed even if retried. The Response Failure indicates that a catastrophic error has occurred and the device should disable its PRI and not issue any more Page Request till it is re-enabled. The Success response indicates that the pages have been made resident and the device should retry its ATS request. 

A page queue overflow is not a catastrophic even that requires disabling of the PRI. The request that triggered the page fault may be recoverable - we just are not able to hand it to the OS. In this case the IOMMU sends a Success response. This tells the device it should retry the ATS translation request - which will obviously fail again - but on this failure the device will resend the Page Request. Hopefully the OS has fixed the overflow condition by now and this resent Page Request will reach the OS.

wangyongzhen0322, 2023-10-28:
I see. Thank you for your answer.

closed 2023-10-28 by wangyongzhen0322