riscv-non-isa/riscv-iommu #263: Can you tell me exactly what the further records refer to?

wangyongzhen0322, 2023-10-27:
I have a question that the term "further records" refers to additional records related to the same function and originating from the same device or there may be no specific requirements for these records.
>  If the IOMMU encounters a fault in accessing the fault-queue memory, the
IOMMU sets the fault-queue memory access fault (fqmf) bit in fqcsr. While either error bit is set in
fqcsr, the IOMMU discards the record that led to the fault and all further fault records.

ved-rivos, 2023-10-27:
I this context, "further" refers to additional or continuing actions. The IOMMU will cease processing and discard all subsequent fault records until the condition is resolved.

wangyongzhen0322, 2023-10-28:
So when the error happens， what will the software do? The description in Section 6.5 Guidelines for handling interrupts from IOMMU is excessively abstract for my comprehension.

ved-rivos, 2023-10-28:
Section 6.5 provides guidelines for software interactions with the IOMMU hardware. However, the section 6 is not intended to dictate policies about what software may want to do. A fault queue overflow is usually not fatal to the operation of the IOMMU and the faulting transaction itself has been provided a protocol specific response. In such case an IOMMU driver may just log the event but not do much more - for example [see](https://elixir.bootlin.com/linux/latest/source/drivers/iommu/arm/arm-smmu-v3/arm-smmu-v3.c#L1590) - but the IOMMU specification does not preclude software from being more clever and doing something different.

ved-rivos, 2023-11-01:
Thanks for raising this question @wangyongzhen0322. Hope that was helpful. If you have any more questions or concerns in the future, please don't hesitate to ask. Closing this issue now

closed 2023-11-01 by ved-rivos