riscv-non-isa/riscv-iommu #265: it seems that the fault queue overflow flag in the reference model lake a condition?

baimengwei, 2023-11-02:
https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_faults.c#L124C30-L124C30
if the fqt is 1023 and fqh is 0, and the max queue size is 1024, then the condition will be 1023 == -1, which seems incorrect?
```
    if ( fqt == (fqh - 1) ) {
        g_reg_file.fqcsr.fqof = 1;
        generate_interrupt(FAULT_QUEUE);
        return;
    }
```

baimengwei, 2024-02-05:
if the fqt is 1023 and fqh is 0, and the max queue size is 1024, is this a full condition, which only can store 1023 record item, rather than 1024 item?

baimengwei, 2024-02-05:
If c model changes the full condition, should the specification change synchronous?
https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_registers.adoc#fault-queue-csr-fqcsr

> The fault-queue-overflow bit is set to 1 if the IOMMU needs to queue a fault record but the fault-queue is full (i.e., fqt == fqh - 1).
> 
> The fault-record is discarded and no more fault records are generated until software clears fqof by writing 1 to the bit.

ved-rivos, 2024-02-05:
> if the fqt is 1023 and fqh is 0, and the max queue size is 1024, is this a full condition, which only can store 1023 record item, rather than 1024 item?
Yes. Please see Chapter 3 -  "A queue is empty if the head is equal to the tail. A queue is full if the tail is the head minus one. The head and tail wrap around when they reach the end of the circular buffer."

baimengwei, 2024-02-06:
> Yes. Please see Chapter 3 - "A queue is empty if the head is equal to the tail. A queue is full if the tail is the head minus one. The head and tail wrap around when they reach the end of the [...]

Thanks

closed 2024-02-06 by baimengwei

yanhe234, 2024-10-23:
> > 如果 fqt 为 1023 且 fqh 为 0，并且最大队列大小为 1024，这是否是完整情况，只能存储 1023 个记录项，而不是 1024 个记录项？
> 
> 请参见第3章-“如果队列头等于队列尾，则队列为空。如果队列尾等于队列头减一，则队列为满。当队列头和队列尾到达循环插座时，它们会回绕。”

Sir, why didn't RV refer to the implementation of arm's transposition row? In extreme cases, the size relationship between fqt and fqh may not be what we want. If we implement the highest bit as the transposition row during implementation, will there be any problems? I'm not sure what significant impact this has? I understand that it's not possible. Looking forward to your reply

ved-rivos, 2024-10-23:
> In extreme cases, the size relationship between fqt and fqh may not be what we want.

Please explain more 

> If we implement the highest bit as the transposition row during implementation, will there be any problems? 

Such an implementation will not be compliant with RISC-V specification and will not be compatible with the RISC-V iommu driver.

yanhe234, 2024-10-23:
> > In extreme cases, the size relationship between fqt and fqh may not be what we want.
> 
> Please explain more
> 

I understand that it is possible that xqt is greater than xqh, for example, fqt>fqh. For example, if production errors are faster than consumption errors, queue is a loop like structure, so it is possible that fqt is greater than fqh. The other two also have similar situations. If there is an error, thank you for correcting it. Thank you