riscv-non-isa/riscv-iommu #268: How does the IO-bridge handle ATS translation requests when the address is a virtual interrupt file address?

18772820305, 2023-11-06:
If an ATS translation request, after being translated by the IOMMU, finds that the requested address A is a virtual interrupt file address, and upon obtaining the MSIPTE, it is determined to be a PTE in MRIF mode. Then, the IOMMU will return information such as is_msi/is_mrif/mrif_nid/mrif_addr to the IO-bridge. How will the IO-bridge handle these contents after retrieval?

ved-rivos, 2023-11-06:
Please see the handling specified in section 2.6.

If the translation request has an address determined to be an MSI address 
but the MSI PTE is configured in MRIF mode then a Success response is
generated with R, W, and U bit set to 1. The U bit being set to 1 in the 
response instructs the device that it must only use Untranslated requests
to access the implied 4 KiB memory range.

When a MSI PTE is configured in MRIF mode, a MSI write with data value `D`
requires the IOMMU to set the interrupt-pending bit for interrupt identity `D`
in the MRIF. A translation request from a device to a GPA that is mapped
through a MRIF mode MSI PTE is not eligible to receive a translated address.
This is accomplished by setting "Untranslated Access Only" (U) field of the
returned response to 1.

18772820305, 2023-11-06:
If the translation request has an address determined to be an MSI address ，and the MSI PTE is configured in MRIF mode then a Success response is generated by IO-bridge. Is it useful for the IOMMU to return NID and NPPN to the IO-bridge?

ved-rivos, 2023-11-06:
> Is it useful for the IOMMU to return NID and NPPN to the IO-bridge?

The contents of the MSI PTE are not useful as they are not included in the response to the device.

ved-rivos, 2023-11-07:
I hope that addressed the question @18772820305. Closing this issue. Please feel free to ask if there are further questions.

closed 2023-11-07 by ved-rivos