riscv-non-isa/riscv-iommu #273: First Stage Translation Clarification

just-for-fun-too, 2023-11-16:
On Page 31, Section 2.2.3 the specification makes the following statement regarding the PC.fsc.PPN which holds the PPN of the root page of a first-stage page table:

"When second-stage address translation is not Bare, the PC.fsc.PPN field holds a guest PPN of the
root of a first-stage page table. Addresses of the _first-stage page table entries_ are then converted by
guest physical address translation process, as controlled by the DC.iohgatp, into a supervisor
physical address."

On Page 25, Section 2.13 the specification makes the following statement regarding the iosatp.PPN:

"When second-stage address translation is not Bare, the iosatp.PPN is a guest PPN. The GPA of the
_root page_ is then converted by guest physical address translation process, as controlled by the
iohgatp, into a supervisor physical address"

My understanding would be that in all cases the first stage addresses (whether you go directly from the DC or through the PC), if the second-stage is not Bare, should be translated. This should be for the first-stage root page address and the subsequent page table entries. However on page 25 it just calls out the root page address and says nothing about the page table entries, while on page 31 it doesn't say anything about the root page address but calls out the page table entries. 

How does this work? Thanks

ved-rivos, 2023-11-17:
The text in 2.1.3 does not further state that the addresses of the PTEs are also converted. However, the process outlined in section 2.3 provides that. We can unify the two text to avoid confusion.

just-for-fun-too, 2023-11-19:
I presume what you are referring to:

In Section 2.3  (Process to translate an IOVA) step #17 there is a reference to "Two-stage Address Translation" of the RISC V Privileged specification (which is section 8.5 in that spec.) 

In section 8.5 of the Privileged spec there is the following statement:

" For any virtual memory access, the original virtual address is converted in the first
stage by _VS-level address translation_"

However no where in the Privileged spec (at least I couldn't find it, I am using 2021203 version) is there a clear description of _VS-level address translation_. I also couldn't find any reference, in there, that the first stage base address and PTEs have to go through a guest address translation if second stage is not Bare (or 0.) In any case I feel any clarification on this issue in either of the specs will be helpful to everyone.

I am assuming based on your answer that of course the first stage address (base address and PTE generated addresses) have to go through a second-stage translation if the second-stage is not Bare.

ved-rivos, 2023-11-19:
> I also couldn't find any reference, in there, that the first stage base address and PTEs have to go through a guest address translation if second stage is not Bare (or 0.)

Memory accesses to read or write page tables are "implicit memory accesses". With that context please refer to following sections:
1. Section 9.5. "When V=1, memory accesses that would normally bypass address translation are subject to G-stage address translation alone. This includes memory accesses made in support of VS-stage address translation, such as reads and writes of VS-level page tables."
2. Table 9.12 and description of trap instruction register contents for guest-page faults. "For guest-page faults, the trap instruction register is written with a special pseudo-instruction value if: (a) the fault is caused by an implicit memory access for VS-stage address translation, and (b) a nonzero value (the faulting guest physical address) is written to `mtval2` or `htval`. If both conditions are met, the value written to `mtinst` or `htinst` must be taken from Table 9.12; zero is not allowed."
3. section 9.2.8 - "A guest-page fault may arise due to an implicit memory access during first-stage (VS-stage) address
translation, in which case a guest physical address written to htval is that of the implicit memory access that faulted—for example, the address of a VS-level page table entry that could not be read"

>Is there a clear description of VS-level address translation.

The VS-stage address translation is identical to single-stage address translation. The operating system that executes in a guest is usually not aware of the fact that it executes in a VM and sets up its page tables as-if it was not executing in a virtual machine. The G-stage is then used to translate its memory accesses.

just-for-fun-too, 2023-11-19:
Thank you! As you pointed out with your three references, its all there, just somewhat spread out which you covered completely in your response.

closed 2023-11-19 by just-for-fun-too