riscv-non-isa/riscv-iommu #277: has SIOV support for RISC-V？

runninglinuxkernel, 2024-01-02:
I saw RISC-V IOMMU has supported PASID, but I want to know is it similar functions or spec like Intel SIOV in RISC-V world?

ved-rivos, 2024-01-02:
奔跑吧Linux社区 wrote:
> I saw RISC-V IOMMU has supported PASID, but I want to know
> is it similar functions or spec like Intel SIOV in RISC-V
> world?

The RISC-V IOMMU compatible with the PCIe specifications. It
does not fully support the scalable I/O virtualization.

SIOV includes a set of rules/requiremnts to be implemented by
the device and the host.

For the host, this inlcudes:
  1.Supporting the PCIe PASID TLP prefix
  2.PASID granular address translation
  3.Direct delivery of interrupts to VMs
  4.DMWr virtualization

The RISC-V IOMMU specification supports 1 and 3. 

The specification does not support selecting a G-stage page
table based on PASID as required for #2. The G-stage page
tables are selected based on the routing ID in the TLP
(seg/bus/dev/fn #). The implication of this is that all ADIs
hosted by a PCIe function need to be associated with a
single VM. 

There is no architecturally defined mechanism in RISC-V to
generate a DMWr PCIe transaction.

regards
ved

ved-rivos, 2024-01-04:
Please feel free to ask if there are further questions.

closed 2024-01-04 by ved-rivos