riscv-non-isa/riscv-iommu #278: How to invalidate the MSI page table cache？Are the two addresses A in Figure 1 and Figure 2 the same?

18772820305, 2024-01-04:
1.
[image]

2.
[image]

runninglinuxkernel, 2024-01-04:
The instructions of invalid MSI page table have provided at chapter 6.3.3.
I think the address A in those two places are the same, GPA. 
MSI page table doing the address translation of Guest interrupt file from GPA  to SPA.

ved-rivos, 2024-01-05:
@18772820305 please let me know if there are further questions.

18772820305, 2024-01-05:
Can the ADDR operand of IOTINVAL.GVMA command be a GVA?

ved-rivos, 2024-01-05:
Please see Table 10 row 3

ved-rivos, 2024-01-13:
Closing this issue. Hope that addressed the question. Please feel free to ask if there are further questions.

closed 2024-01-13 by ved-rivos