riscv-non-isa/riscv-iommu #279: remove "PDT" word

runninglinuxkernel, 2024-01-05:
In chapter 3.1.1,

IOMMU operations cause implicit reads to PDT, first-stage and second-stage page tables. To reduce
latency of such reads, the IOMMU may cache entries from the first-stage and/or second-stage page
tables in the IOMMU-address-translation-cache (IOATC).

This chapter just talk about invalidate the S1 and S2 page table cache, so for the first sentence, should we remove the word "PDT" avoid confusing?

ved-rivos, 2024-01-05:
The paragraph discusses the implicit reads to PDT to draw attention explicitly to the fact that when second-stage is not Bare, the `PPN`s held in the non-leaf PDT entries and in the `pdtp` are guest PPNs. They are translated by the second-stage page table to determine the SPA to access the non-leaf PDT entries and the `PC`. This process could lead to the second-stage address-translation caches being updated.

runninglinuxkernel, 2024-01-06:
yes, your explanation is much better the first sentence of chapter 3.1.1 which is confusing me or other readers.

ved-rivos, 2024-01-06:
This has been explained in detail before - e.g. section 2.3.2,

runninglinuxkernel, 2024-01-06:
Thanks a lot!

closed 2024-01-06 by runninglinuxkernel