riscv-non-isa/riscv-iommu #280: how to understand the PR/PW bit in IOFENCE command.

runninglinuxkernel, 2024-01-14:
In chapter 3.1.2, it has one sentence talk about the PR/PR bit like below:

```
The PR bit, when set to 1, can be used to request that the IOMMU ensure that all previous read
requests from devices that have already been processed by the IOMMU be committed to a global
ordering point...
```

how to understand the words "all previous read requests from devices", is it related to Commands submitted to CQ or 
all of normal read from device?

In chapter 6.3， it has one sentence:

```
The ordering on stores provided by FENCE
instructions and the acquire/ release bits on atomic instructions also orders the data structure
updates associated with those stores as observed by IOMMU.
```

Does this sentence want to say the fence instructions in RISC-V also affect the orders of the IOFENCE commands?

runninglinuxkernel, 2024-01-14:
> > how to understand the words "all previous read requests from devices"
> 
> The IOFENCE.C command may be used to order memory accesses from I/O devices connected to the IOMMU as viewed by the IOMMU, other RISC-V harts, and external devices or co-processors. All reads from a device that are previous to the IOFENCE.C command are ordered by the PR bit.
> 
> > Does this sentence want to say the fence instructions in RISC-V also affect the orders of the IOFENCE commands?
> 
> No it does not. That section is not related to or relevant to ordering memory accesses from I/O devices.

some code like below:
```
sw a1，(s1)
IOFENCE A
sw a2，(s2)
IOFENCE B
````

is it possible that execute the "IOFENCE B" firstly than the "IOFENCE B" by CPU speculation?

ved-rivos, 2024-01-14:
The `IOFENCE.C` is not a CPU instruction. The `IOFENCE.B` is a command executed by the IOMMU. These commands are submitted to the IOMMU through the command queue. A `IOFENCE.C` command completion, as determined by cqh advancing past the index of the `IOFENCE.C` command in the CQ, guarantees that all previous commands fetched from the CQ have been completed and committed.

ved-rivos, 2024-01-16:
Hope that cleared up the understanding of the IOFENCE command. Please let me know if you have further questions.

closed 2024-01-16 by ved-rivos