riscv-non-isa/riscv-iommu #281: How does IOMMU ensure that it does not receive outdated PCIe inbound traffic after completing TLB invalidation?

18772820305, 2024-01-25:


ved-rivos, 2024-01-26:
How do you define outdated in this context?

ved-rivos, 2024-02-05:
Closing this issue. Please feel free to ask if there are further questions.

closed 2024-02-05 by ved-rivos

yanhe234, 2024-10-09:
I think the author of this question wants to ask: When PCIe devices initiate address access, they use the old address A. However, if the invalid command is later than the access to address A initiated by PCIe, will IOMMU translate address A into the wrong address？

ved-rivos, 2024-10-09:
When translation for an address A is changed by modifying a PTE, then the IOMMU may either use the old PTE contents or the new PTE contents till an invalidation is issued and completed by the IOMMU.

yanhe234, 2024-10-09:
Let's assume a very extreme scenario, when a PCIe device initiates a translation of address A, and then for some reason the software needs to invalidate the corresponding PTE of address A. In this case, logically, I think the translation of address A initiated by the PCIe device should expect to get the old translation result, but it may get a new translation result. Is there a problem with this?

ved-rivos, 2024-10-10:
When software changes the translation for an address A, then software expects that for the short duration where the invalidation has not been completed, hardware may use either the old translation or the new translation. Effectively, software cannot use the memory corresponding to old translation till the invalidation is completed for any other purposes. This is expected behavior of address translation caches and TLBs and software comprehends that. There is no problem with that.

yanhe234, 2024-10-10:
Very clear, thank you very much

yanhe234, 2024-10-15:
Hello sir, I have a question. When the page req queue is full and a stop marker message arrives, there is no need to return a response. However, if L=0, there is no need to respond. Only when L=1, there is a need to respond with a successful response. Why is there a need to return a successful response? Is the corresponding content also the content of the incoming page request?

ved-rivos, 2024-10-15:
The mechanism is designed to allow for a graceful recovery from the queue full condition. Sending a success response to the device indicates that the device may now retry the ATS translation request that originally lead to the page fault. In the meanwhile a queue full interrupt notification has also been provided to the host software. This retried ATS translation request will likely encounter a page fault again leading the device to generate another set of page requests. If software has been able to correct the queue full condition then these page requests will get delivered to software. If not the process continues. The page group response contains the page request group index from the page request and the response code.

yanhe234, 2024-10-15:
Very clear, thank you