riscv-non-isa/riscv-iommu #284: Sorry,how to configure the value of capabilities.PAS?

augsug, 2024-03-13:
PAS is a 6-bits field in capabilities,which means Physical Address Size supported by the IOMMU.But how can i configure this filed?

[image]

ved-rivos, 2024-03-14:
The `capabilities` register is used to report the capabilities of the IOMMU hardware. This is a read-only register used for enumeration and is not a configuration register. So the `PAS` (and other hardware capabilities) are not software configurable. Some hardware implementations, such as soft-IPs, may have build time configurations for the hardware and may also support mechanisms like straps/fuses to configure the hardware  at implementation time.

ved-rivos, 2024-03-18:
I hope that addressed the question. Please feel free to ask if there are further questions.

closed 2024-03-18 by ved-rivos

augsug, 2024-03-25:
I got u.  if i set this field in 6'b0000_01，what does this mean?Physical Address Size is 4KiB or something ? 
look forward to your advice:)

ved-rivos, 2024-03-25:
@augsug - a value of 6'b0000_01 implies 1-bit and a value of 6'b1110_00 implies 56-bit.

viktoryou, 2025-08-15:
Seems that a value of 0 would be meaningless or unspecified, right?