riscv-non-isa/riscv-iommu #291: Ambiguity and conflict of IOMMU and PCIe ATS translation permission result

zetalog, 2024-04-19:
When PCIe ATS sends translation requests to IOMMU, IOMMU is supposed to return RWX permissions for the request.

According to our compliance test, there is an ambiguity in the specification, or known as a conflict between IOMMU specification and the de-facto standard PCIe ATS implementations (ARM DTI specification seems to have been adopted by the IP vendors).

### IOMMU specification - 2.6. PCIe ATS translation request handling

When translation could not be completed due to the following causes a Success Response with R and W bits set to 0 is generated. (cause 12/13/15/20/21/23/266/262).
If the translation request has a PASID with "Privilege Mode Requested" field set to 0, or the request does not have a PASID then the request does not target privileged memory. If the U-bit that indicates if the memory is accessible to user mode is 0 then a Success response with R and W bits set to 0 is generated.
If the translation request has a PASID with "Privilege Mode Requested" field set to 1, then the request targets privileged memory. If the U-bit that indicates if the page is accessible to user mode is 1 and the SUM bit in the ta field of the process-context is 0 then a Success response with R and W bits set to 0 is generated.

If the translation could be successfully completed but the requested permissions are not present (Execute requested but no execute permission; no-write not requested and no write permission; no read permission) then a Success response is returned with the denied permission (R, W or X) set to 0 and **the other permission bits set to the value determined from the page tables.**

We can also see the IOMMU reference model implemented in this way.

### DTI specification - 

5.2.2 DTI_ATS_TRANS_RESP

ALLOW_X, bit [66] 
    This bit indicates permissions for instruction reads.
    0    Not permitted
    1    Permitted
    When the value of ALLOW_R is 0, this bit must be 0. When the value of InD in the DTI_ATS_TRANS_REQ translation request message was 0, this bit must be 0.

IMO, we should change the IOMMU specification to be compliant with de-facto standards accommodated by ecosystem IP vendors. And there is no benefit returning page table attributes to the requester while the permission is not requested.

ved-rivos, 2024-04-19:
You are right that the X bit should be qualified by the Execute Requested. Update made in this commit [for](https://github.com/riscv-non-isa/riscv-iommu/pull/243/commits/b61302c0b85aec75683945ced70581edd05ff93e).

zetalog, 2024-04-22:
One more performance issue noticed.
Let me use a new thread.

closed 2024-04-22 by ved-rivos