riscv-non-isa/riscv-iommu #309: Question related to Bypass/Global bits in ATS_TRANS_RESP

zetalog, 2024-05-07:
Our verification guys noticed a difference between the refence model and the model they developed based on SPEC:

```
        // only use Untranslated requests to access the implied 4 KiB memory range
        ...
        rsp_msg->trsp.Global = vs_pte.G;
        ...
```

ATS response uses VS PTE.G which doesn't reflect to the following description:

> **5.2.2 DTI_ATS_TRANS_RESP**
> BYPASS, bit [17] 
> This field indicates that translation for this StreamID is bypassed.
>  0 Normal translation
>  1 Translation bypassed
> When the value of this field is 1, the VA and the PA of the translation are the same. This bit must be 0 if the value of IA in the translation request is greater than the range shown in the OAS field of the DTI_ATS_CONDIS_ACK message that was received during the connection sequence.

And there is no  Global bit defined:
[image]

in this revision:
[image]

Now we are wiring these bits to 0s both in our design and the verification model.

Is this a compatibility issue in the reference model?

ved-rivos, 2024-05-07:
Mapping to protocols such as DTI should follow the rules provided by those specifications. The IOMMU specification follows the PCIe specification and any such mapping to DTI/LTI or other implementations is implementation defined.

zetalog, 2024-05-08:
Sorry, my mistake.
This is the one from the recent DTI specification:

[image]

ved-rivos, 2024-05-08:
The previous picture you posted is for DTI TBU. As noted above, the mapping to implementation specific buses like DTI is implementation specific.

zetalog, 2024-05-08:
From PCIe 5.0, there is no Bypass field in CPL:

And can see Global bit in Table 10-3:

```
Global Mapping - If this bit is Set, the ATC is permitted to cache this mapping entry in all PASIDs. If Clear, the ATC is
permitted to cache this mapping entry only in the PASID associated with the requesting PASID. This bit may only be Set
if the associated Translation Request had a PASID TLP Prefix.
```

ved-rivos, 2024-05-08:
> From PCIe 5.0, there is no Bypass field in CPL:

Bypass is a DTI concept and as you noted not a PCIe concept.

zetalog, 2024-05-08:
According to our PCIe guys, PCIe IPs containing "Global" bit follow DTI 2.0 and early DTI 3.0 spec.

ved-rivos, 2024-05-08:
Please check if you see any reference to DTI in PCIe specification.

zetalog, 2024-05-08:
> The previous picture you posted is for DTI TBU. As noted above, the mapping to implementation specific buses like DTI is implementation specific.

Shall the implementation dependency be mentione for this paragraph?

**2.6. PCIe ATS translation request handling**

• _Global field is set to the value determined from the first-stage page tables if translation could be successfully completed and the request had a PASID present_. In all other cases, including MSI address translations, this field is set to 0.

zetalog, 2024-05-08:
> Please check if you see any reference to DTI in PCIe specification.

There is no reference of DTI can be found in PCIe spec, just can see reference of DTI in the IP data book.

ved-rivos, 2024-05-08:
Such an IP using DTI always receives Global as 0.  It is equivalent to such an IP ignoring the Global bit - which is allowed by PCIe - this only results in multiple translations being requested for the same Untranslated Address under different PASIDs - see 10.2.3.8.

zetalog, 2024-05-08:
> Such an IP using DTI always receives Global as 0. It is equivalent to such an IP ignoring the Global bit - which is allowed by PCIe - this only results in multiple translations being requested for the same Untranslated Address under different PASIDs - see 10.2.3.8.

Is that means if an implementation doesn't support ATS.INVAL G field, it can choose to respond G=0 to ATS translation request?

ved-rivos, 2024-05-09:
Firstly, I would like to request you to not reproduce information from the PCIe specification that is licensed to you in this GitHub. If you need to refer to a section then providing a section number is sufficient. Would want to avoid any copyright, compliance, antitrust, etc. issues. An implementation that always responds to endpoint with the G bit set to 0 may not run into issues.

zetalog, 2024-05-09:
> Firstly, I would like to request you to not reproduce information from the PCIe specification that is licensed to you in this GitHub. If you need to refer to a section then providing a section [...]

Sure, just sorted all of them out. Thanks for the reminds.

> The previous picture you posted is for DTI TBU. As noted above, the mapping to implementation specific buses like DTI is implementation specific.

My question is still about the specification/reference model clarification.
From the RISCV IOMMU specification, we cannot infer ATS handler are allowed to respond G=0, while PCIe spec permits.

ved-rivos, 2024-05-09:
The PCIe specification permits ignoring the response from the IOMMU. The RISC-V IOMMU specification is compliant with the PCIe specification and no further note is required.

zetalog, 2024-05-10:
OK, thanks, marking it closed.

closed 2024-05-10 by zetalog

space-mit, 2024-05-20:
One more thing I'd like to mention is:

For an implementation that is based on DTI. "Bypass" is a necessary field for ATS response, it combines with "Range" field. Upon accepting such an translation response, the DMA master side (e.x., PCIe RC) DTI bridge is responsible for generating "output address" using "input address" and "range", "bypass" fields, with lower bits indicated by "bypass/range" coming from the "input address" in translation request and higher bits coming from "output address" in translation response.

As such, "Bypass" is also used in ATS translations.

However, the current reference model seems like a TLP base model, cares less about DTI ATS interfacing.
Probably should move all of those ATS bridge side stuffs to implementation specific or test bench specific, and not be a part of IOMMU model.

zetalog, 2024-05-20:
One more thing I'd like to mention is:

For an implementation that is based on DTI. "Bypass" is a necessary field for ATS response, it combines with "Range" field. Upon accepting such an translation response, the DMA master side (e.x., PCIe RC) DTI bridge is responsible for generating "output address" using "input address" and "range", "bypass" fields, with lower bits indicated by "bypass/range" coming from the "input address" in translation request and higher bits coming from "output address" in translation response.

As such, "Bypass" is also used in ATS translations.

However, the current reference model seems like a TLP base model, cares less about DTI ATS interfacing.
Probably should move all of those ATS bridge side stuffs to implementation specific or test bench specific, and not be a part of IOMMU model.