riscv-non-isa/riscv-iommu #311: Question about the description of t2gpa usage scenario

viktoryou, 2024-05-08:
From my understanding, for a PCIe ATS Translation Request, the usage scenario of t2gpa when enabled would typically be like:
* firstly, untranslated request from device triggers IOMMU to perform the two-stage address translation, and GPA returns and gets cached in device ATC,
* then translated request from device accesses IOMMU with GPA to continue the second-stage translation, and SPA returns to device.

It seems like t2gpa means _translating to gpa_.  Is the pattern above a fixed step-by-step usage flow? In other words, could we say it is expected a untranslated request would be immediately followed by a related translated request(with same gpa)? If this is the most common case, then I guess the translation cache is important for the second stage when t2gpa is enabled.

ved-rivos, 2024-05-08:
> firstly, untranslated request from device triggers IOMMU to perform the two-stage address translation, and GPA returns and gets cached in device ATC,

Its only applicable to ATS Translation requests - not untranslated request. On a ATS Translation request, the IOMMU does an IOVA -> GPA -> SPA translation. However, it only returns the GPA and not the SPA in the completion to the device. The translation from GPA->SPA is done to ensure there are no faults and to determine accurate permissions. 

A translation request in this usage would be followed by a translated requested using the translated address - a GPA in this configuration. The GPA in the translated request is then again translated to an SPA and the permissions verified.

viktoryou, 2024-05-08:
That's clear. Thanks a lot.

closed 2024-05-08 by viktoryou