riscv-non-isa/riscv-iommu #312: Question about device ID validity detection in PRI request

viktoryou, 2024-05-10:
From IOMMU model, I found the validity detection of device ID in processing a "Page Request" or "Stop Marker" message is missing ( step 3 ~ step 5 in iommu_translate.c). I guess this is a bug in iommu_atc.c since it skips the detection before locating device context.

Additionally, for the validity detection of device ID in an PCIe ATS transaction, under a DTI-ATS protocol for example,  what if the higher part of SID (bit[31:25]) illegally set from DTI_ATS_TRANS_REQ or DTI_ATS_PAGE_REQ? I think we may not guarantee that it is impossible. I found some description from the spec,
> The IOMMU does not validate the authenticity of the hardware IDs provided by the IO bridge. 
The IO bridge and/or the root ports must include suitable mechanisms to authenticate the hardware IDs.

For an illegal scenario above, what's the expected way to cope with? It seems that IOMMU would only use the max supported device ID (24 bit at maximum as defined), regardless of whether higher bits in device ID being legal.

ved-rivos, 2024-05-10:
> I guess this is a bug in iommu_atc.c since it skips the detection before locating device context.

Thanks for spotting that. 

> For an illegal scenario above, what's the expected way to cope with? It seems that IOMMU would only use the max supported device ID (24 bit at maximum as defined), regardless of whether higher bits in device ID being legal.

The rest of that note has more guidance. 

> The IO bridge and/or the root ports must include suitable mechanisms to authenticate the hardware IDs. In some SOCs this may be trivially achieved as a property of the devices being integrated into the SOC and their IDs being immutable. For PCIe, for example, the PCIe defined Access Control Services (ACS) Source Validation capabilities may be used to authenticate the hardware IDs. Other implementation-specific methods in the IO bridge may be provided to perform such authentication.

viktoryou, 2024-05-11:
Could we say that authentication of hardware IDs is invisible to IOMMU, and validity is guaranteed by either IO bridge or PCIe Root Port or both?

ved-rivos, 2024-05-11:
Right.

closed 2024-05-11 by viktoryou