riscv-non-isa/riscv-iommu #316: Should the set of both AV and WSI in iofence command be illegal

viktoryou, 2024-05-11:
When an IOFENCE.C command completes,  AV or WSI could be configured for the indication. The WSI bit when set to 1 causes a wired-interrupt and AV=1 indicates a 4-byte store. However, there is no restriction whether AV and WSI could be set at the same time.

When fctl.wsi=0, setting WSI in IOFENCE.C is illegal and we could use AV bit to indicate the IOFENCE.C completion.
When fctl.wsi=1, when AV and WSI bit are set together, their behavior would both be valid.  

Taking note of this description from spec,
> Software may configure the ADDR[63:2] command operand to specify the address of the seteipnum_le/seteipnum_be register in an IMSIC to cause an external interrupt notification on IOFENCE.C completion. Alternatively, software may program ADDR[63:2] to a memory location and use IOFENCE.C to set a flag in memory indicating command completion.

If the software follows the first option, then it seems a MSI write as well as a WSI notification would both happen. This is a bit stange.
If the software takes alternative option, this indication looks redundant since wired-interrupt is provided as well.

From my understanding, I assume that AV and WSI bit are not expected to be set together.  Further, if this is right, should the set of both AV and WSI in iofence command be illegal?

ved-rivos, 2024-05-11:
The `AV` and `WSI` may usually not be set together but it is not illegal for software to request that.

closed 2024-05-11 by viktoryou

viktoryou, 2024-05-17:
I may need to confirm one point.
When AV and WSI are both set, their behavior would both be valid (though it maybe unexpected). Under this circumstance, should any priority between them or both be allowed?

viktoryou, 2024-05-19:
Thanks a lot.