riscv-non-isa/riscv-iommu #320: Should the hpm counter still count when a fault is checked?

wangyongzhen0322, 2024-05-20:
For example, if a fault occurs while locating the DC, whether the register that counts the Walks in the Device Directory should be counted. If the IOMMU receives an ATS Translation request when ddtp.iommu_mode==Bare, whether the register that counts ATS Translation requests should be counted.

ved-rivos, 2024-05-20:
The performance monitoring counter that counts walks in device directory would increment by one on each read from the device directory table. When the mode is `Bare`, there are no reads from the device directory table and the counter will not count. A counter that counts ATS translation requests will count by one on each such request irrespective of whether the request leads to a failure. Effective, the counters count events that they are programmed to count.

wangyongzhen0322, 2024-05-20:
I got it. Thanks for your explanation

closed 2024-05-20 by wangyongzhen0322