riscv-non-isa/riscv-iommu #321: Questions about ATS translation request and ATS Page Request handling

wangyongzhen0322, 2024-05-20:
1.  Whether the DTF affects the following logged in the fault queue
    > ATS [1] translation requests that encounter a configuration error results in a Completer Abort (CA)
response to the requester. The following cause codes belong to this category:
• Instruction access fault (cause = 1)
• Read access fault (cause = 5)
• Write/AMO access fault (cause = 7)
• MSI PTE load access fault (cause = 261)
• MSI PTE misconfigured (cause = 263)
• PDT entry load access fault (cause = 265)
• PDT entry misconfigured (cause = 267)

2.  Whether the following status cause a fault to log in the fault queue? The conditons where no faults are logged in the fault queue for PCIe ATS "Page Request" messages don't include both the following status
    > The status is set to Invalid Request if the following faults are encountered:
• ddtp.iommu_mode is Bare
• EN_PRI is set to 0

ved-rivos, 2024-05-20:
> Whether the DTF affects the following logged in the fault queue

Yes it does. Please also see table 11.

> Whether the following status cause a fault to log in the fault queue? The conditons where no faults are logged in the fault queue for PCIe ATS "Page Request" messages don't include both the following status

Yes, they will. Please also see section 4.2 that states "If a fault condition prevents locating a valid device context then the DTF value assumed for reporting such faults is 0."

wangyongzhen0322, 2024-05-20:
> Yes, they will. Please also see section 4.2 that states "If a fault condition prevents locating a valid device context then the DTF value assumed for reporting such faults is 0."
So what is the cause of each of them?I think that when ddtp.iommu_mode is Bare,report"Transaction type disallowed" (cause = 260);and when EN_PRI is set to 0, report "DDT entry misconfigured" (cause = 259).And the ttyp should be "PCIe Message Request"(ttyp ==9).Is that right?

ved-rivos, 2024-05-20:
The cause is 260 for both of these and the transaction type is message request.

wangyongzhen0322, 2024-05-20:
I also have another question. For the PRI fault report, the software can get the relevant information through the interruption caused by fip or pip, except when pqen=0 or pqon=0. I can understand why a fault report is not generated to log in fault queue when page-request queue encountered a memory access fault (pqcsr.pqmf == 1) or "Page Request" could not be queued due to the page-request queue being full (pqt == pqh - 1) or had a overflow (pqcsr.pqof == 1). That's because pqof and pqmf themselves generate an interrupt. Why there is no fault report logged in fault queue when pqen==0 or pqon==0?

ved-rivos, 2024-05-20:
> Why there is no fault report logged in fault queue when pqen==0 or pqon==0?

When software has just disabled the page requests queue then it is not useful to remind software that it has disabled the page request queue.

wangyongzhen0322, 2024-05-20:
Thanks for your answer.

closed 2024-05-20 by wangyongzhen0322

yanhe234, 2025-02-24:
Hello，Sir.   The scenario you mentioned involves potential cross-write issues when multiple program groups (PRG groups) are executing in parallel. Specifically, when one PRG group sends its last (Last) content, while another PRG group's data has not been fully written (i.e., "incomplete"), how would the software handle such an incomplete PRG group?

yanhe234, 2025-02-24:
Of course, the previous question refers to the scenario where the **PQ is not full**.

ved-rivos, 2025-02-24:
If software has disable the page request queue while there are devices generating page reuests then in that scenario the page requests received after the software disable the page request queue will not get reported. It’s likely software does not care for those - if disabled the page request queue - or it intends to reset the device or its PRI . The condition of a partially stored page requests group can occur if software invokes such actions

yanhe234, 2025-02-25:
Hello, sir, let me clarify my question scenario. When the Pq queue is not disabled and there is no overflow, and two or more prg groups are interleavedly reporting and writing to memory, when one of the groups receives the final "last page req," the other groups are relatively incomplete. How will the software handle this? Your response did not seem to address my question. I look forward to your reply. Best regards.

ved-rivos, 2025-02-25:
Software will need to wait for the second group to complete . Effectively software has to read each page request from the queue and group them into groups.

yanhe234, 2025-02-25:
very  clear,  sir ,many  thanks

yanhe234, 2025-03-13:
Hello sir, I have a question during my code implementation. Regarding scenarios where the IOMMU hardware automatically replies with an RSP in error conditions, will this RSP only send a failed page response for the last request in a group of requests?

yanhe234, 2025-03-13:
My concern is that encountering a write error while writing page requests may pull PQMF high, potentially causing the loss of several page requests. The corresponding command responses would not address these dropped responses. I look forward to your reply.

ved-rivos, 2025-03-13:
Only the last in the group needs an auto response

yanhe234, 2025-03-13:
> My concern is that encountering a write error while writing page requests may pull PQMF high, potentially causing the loss of several page requests. The corresponding command responses would not address these dropped responses. I look forward to your reply.
My point is that when multiple page requests encounter a write operation to PQMF, the subsequent ones will automatically respond, but the software later handles this error, so the following requests will report normally. In this case, the software discards some data. According to your advice, how can we avoid this situation? Did I describe the issue clearly?

ved-rivos, 2025-03-13:
The reason for causing the auto response is to encourage the device to retry. Software may find  an incomplete group in such cases and they are expected to ignore it - including doing an auto response from sw for if a last request was found. Since the device retries, eventually if software fixes the fault the page requests will land successfully

yanhe234, 2025-03-13:
many  thanks  sir

yanhe234, 2025-03-13:
> My concern is that encountering a write error while writing page requests may pull PQMF high, potentially causing the loss of several page requests. The corresponding command responses would not address these dropped responses. I look forward to your reply.

Because of this, I understand that it will automatically reply with an ACK.

yanhe234, 2025-03-13:
Sir, your answer does not address my question. When I mention "mem fault" here, I am not referring to memory overflow (mem ovf), but rather to a scenario where data was not written to memory due to some reason.

ved-rivos, 2025-03-13:
If the page request queue had a fault - due to an overflow or due to a software bug like locating the page request queue in memory that leads to a memory fault - then software must assume that a lot of page requests were lost. The treatment of a page request queue fault - due to any reason - is the same.

yanhe234, 2025-03-13:
From my understanding, simply modifying the automatic response rules is insufficient here. For page requests within the group that are not the final "L" marker, the system should automatically return failure responses. This approach could reference ARM's specific implementation
**From the software perspective, it assumes a group is complete if L==1 is present in memory. Please correct me if there are any errors in this understanding.**

ved-rivos, 2025-03-13:
When a fault occurs the page requests queue stops storing page requests into the queue. So consider a group with 2 messages - first one with L=0 and second one with L=1. If the first message encounters a fault - an overflow or a memory fault - then it will be discarded and no auto response occurs. The page queue is marked as having a memory fault and is disabled. The second message now sees the page queue as disabled and will also be discarded. However, since L == 1, there is a automatic response to the device. Could you explain your scenario.

yanhe234, 2025-03-13:
**Sir, I apologize for any confusion caused by my limited English. Let me clarify: During the second occurrence of L=1, the software has already handled the memory fault (mf) error and can now write to memory normally. However, at this stage, the software does not perceive a logically complete group, because the initial request was not successfully written to memory.**

yanhe234, 2025-03-13:
many thanks

yanhe234, 2025-04-28:
Hello, sir. Regarding the DTI protocol of ARM when handling page request messages, the IOMMU returns an ATS_Page_RespAck message. I would like to ask: what is the significance of this? Why can’t it simply end by returning ATS_Page_Resp? While adding RespAck does make the process more robust, are there other reasons behind this design?