riscv-non-isa/riscv-iommu #331: Question about access_check or permission_check of PRS message

HDntown, 2024-05-28:
I would like to inquire about the check of PRS-related messages in IOMMU.  In PCIE page request message，if we set W bit，R bit and  E bit(in prefix)，do these bit fields transmitted from PCIe PRS need to follow the checking mechanism defined in iommu_reference_model when they are parsed by the IOMMU?  Just following check.

    *read = ( req->tr.read_writeAMO == READ ) ? 1 : 0;
    *write = ( req->tr.read_writeAMO == WRITE ) ?  1 : 0;

    *exec = ( (*read && req->exec_req &&
                (req->tr.at == ADDR_TYPE_UNTRANSLATED || req->pid_valid)) ) ? 1 : 0;
    *priv = ( req->pid_valid && req->priv_req ) ? S_MODE : U_MODE;
    
    And I'm not sure if these checks are only applied to memory accesses.  PCIE PRS is out of this type，so W bit，R bit and  E bit are  are not suitable for  above check. However, these bit fields will participate in the attribute comparison during the process of software transferring data from the hard disk to the memory in PRS.  Is this right? 
    Thank you!

ved-rivos, 2024-05-28:
> And I'm not sure if these checks are only applied to memory accesses.  

For PCIe, these only apply to Mrd*, MWr, FetchAdd, Swap, and CAS TLPs. These requests do not apply to messages - Msg/MsgD TLPs. Page Request is a message TLP. The IOMMU creates a page-request-queue record on receipt of the page request message but does not do any further checks on the payload of the message.

baimengwei, 2024-05-28:
> For PCIe, these only apply to Mrd*, MWr, FetchAdd, Swap, and CAS TLPs. 
Why not include IO* ?

ved-rivos, 2024-05-28:
IO* requests access the I/O space and not memory. 
PCIe only supports I/O space for compatibility with legacy PCI devices. PCIe devices are not permitted to generate IO* TLPs. They may support completing I/O requests. A root complex may support generating I/O requests for compatibility with legacy devices but is not required to support completing or routing I/O requests.

closed 2024-06-04 by ved-rivos