riscv-non-isa/riscv-iommu #332: Why not add full condition such as DC.tc.PDTV check about handle_page_request ?

baimengwei, 2024-05-29:
EN_PRI is checked after get DC for page request message.

https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_ats.c#L173
```
    if ( DC.tc.EN_PRI == 0 ) {
        // 7. if any of the following conditions hold then stop and report
        //    "Transaction type disallowed" (cause = 260).
        //   * Transaction type is a PCIe "Page Request" Message and `DC.tc.EN_PRI` is 0.
        report_fault(260, PAGE_REQ_MSG_CODE, 0, PCIE_MESSAGE_REQUEST, DC.tc.DTF,
                     device_id, pr->PV, pr->PID, pr->PRIV);
        response_code = INVALID_REQUEST;
        goto send_prgr;
    }
```
However, if a page request message is send with process_id, but the DC.tc.PDTV is 0, why not report fault to fault queue ? it seems that the cause 256-259 cause condition is covered, but cause 260 is not full covered.

ved-rivos, 2024-05-29:
The IOMMU does not need any information from the process context to process the page request. The IOMMU verifies that the device is authorized by SW to generate page requests. The rest of the processing of the page request is in software.

baimengwei, 2024-05-30:
thanks a lot.

baimengwei, 2024-08-20:
> The IOMMU does not need any information from the process context to process the page request. The IOMMU verifies that the device is authorized by SW to generate page requests. The rest of the processing of the page request is in software.

If HW implement checks other fault about cause 260, such as `PDTV`, is it also reasonable ?

ved-rivos, 2024-08-20:
> If HW implement checks other fault about cause 260, such as PDTV, is it also reasonable ?

It will not be compliant to the IOMMU specification - see section 2.7.

baimengwei, 2024-08-21:
thanks a lot.

closed 2024-08-21 by baimengwei