riscv-non-isa/riscv-iommu #345: Fix note icon, UL marker size, dot-leader

wmat, 2024-06-07:
Revert the note admonition icon to the older info circle.
Make unordered lists bullets larger.
Add the TOC dot-leader back in.

ved-rivos, 2024-06-07:
Change codespan font to Jetbrains mono.

ved-rivos, 2024-06-07:
Make page number column in TOC fixed width so the column does not look wavy.
Make Figure number and Table number column in list of figures and list of tables fixed width so that captions line up and dont look wavy. Perhaps something like below:
```
# List of figures settings
lof:
  label_width: '3em'  # Adjust the width for figure labels
  align: left
  caption:
    indent: '4em'  # Ensure captions align consistently

# List of tables settings
lot:
  label_width: '3em'  # Adjust the width for table labels
  align: left
  caption:
    indent: '4em'  # Ensure captions align consistently
```

ved-rivos, 2024-06-07:
Bullets could possible be fixed this way:
```
# List settings
list:
  indent: '$horizontal_rhythm * 1.5'
  item_spacing: '$vertical_rhythm / 2'
  bullet:
    font_size: 'round($base_font_size * 1.5)'
    content: '•'
```

wmat, 2024-06-07:
I'll give that a try.

wmat, 2024-06-07:
Didn't seem to work.

wmat, 2024-06-10:
I've fixed the bullets.  I have it looking like this now:

[image]

ved-rivos, 2024-06-10:
That looks good. Thanks!

ved-rivos, 2024-06-10:
how does the level 2 and level 3 sub-bullet look?

wmat, 2024-06-10:
Here's what all 3 levels look like:

[image]

ved-rivos, 2024-06-10:
The size and shape look good! Could the second level also have color black instead of white.

wmat, 2024-06-10:
How's this:
[image]

ved-rivos, 2024-06-10:
Thanks! Looks good!

wmat, 2024-06-10:
btw, setting the :bibtex-style: apa, which matches the ISA manuals will fix your bibliography alignment issue.

ved-rivos, 2024-06-10:
I think the LaTex ISA manual - before asciidoc coversion - used to use ieee bibtext style. It would be preferred to use ieee style..
[image]

wmat, 2024-06-10:
Ah, OK.  This is a known issue with the ieee style in asciidoc.  I'll see if there's a resolution.

ved-rivos, 2024-06-10:
@wmat - would you be updating the codespan font as well to something like Jetbrains mono?

wmat, 2024-06-10:
Yes, I'll do that as well.

wmat, 2024-06-11:
JetBrainsMono has been added to my docs-resources branch and pointed codespan at it.

As for the Bibliography, it'd be some work to make the ieee formatting match LaTeX.  I know this isn't ideal, but I don't think it's warranted right now.  Can you live with the layout the way it is?  

So what's left for the iommu spec?  The alignment of the LoF and LoT?

ved-rivos, 2024-06-11:
Thanks!

> As for the Bibliography, it'd be some work to make the ieee formatting match LaTeX. I know this isn't ideal, but I don't think it's warranted right now. Can you live with the layout the way it is?

I think I can live with that.

> So what's left for the iommu spec? The alignment of the LoF and LoT?

The two other item I was tracking were
1. "wavy" numbers column in TOC, LoF, and LoT.
2. Lining up the caption text to start from same column in LoF and LoT.

wmat, 2024-06-11:
So if I use JetBrainsMono as the font for lists it's slightly approved, at least for text alignment but I can't seem to get the wavy numbering to go away:

[image]

ved-rivos, 2024-06-11:
The Jetbrains mono was to help solve the weird ligatures that were being previously produced like this "&". Hopefully updating to Jetbrains fixed that

[image]

wmat, 2024-06-11:
Looks like this now:
[image]

ved-rivos, 2024-06-11:
Thanks! This looks nice! The older one rendered "&" and "~" using really odd characters.

wmat, 2024-06-11:
OK, great.  That screen capture was from my local build btw.  I'll make the changes to docs-resources next.

wmat, 2024-06-11:
The GitHub action build is now using all of these fixes in docs-resources.  Should probably close this now and open new issues as you find them.

ved-rivos, 2024-06-11:
One thing I noticed is that the text inside tables is much smaller than the main body text. Was that intentional?

wmat, 2024-06-11:
I thought that was fixed. I’ll look shortly.

Bill Traynor
Documentation Build and Release Engineer
RISC-V International

Join us in Munich, Germany at RISC-V Summit Europe
<https://riscv-europe.org/summit/2024/> from 24-28 June. Be a part of the
new wave of European computing innovation!

wmat, 2024-06-12:
So table_font_size was changed to 11.5 recently and then overwritten back to 9 with the merging of the two themes.  9 seems a bit small to me.  I'm thinking a size of ($base_font_size * 0.9em) would be appropriate.  That is the equivalent to 10.35 pt when base font size is 11.5 pt.

ved-rivos, 2024-06-12:
Thanks! That looked good!

ved-rivos, 2024-06-12:
One small nit if its not too hard to fix. I think the bullets are now a bit too big - I believe they are set to 1.5xbase-font - perhaps 1.25xbase-font is more appropriate. I pasted a google doc compared to the current render.
[image]

wmat, 2024-06-12:
I actually am just using unicode glyphs.  I believe that one is U-26AB.  Does this look better?  I can hardly see the difference.

[image]

wmat, 2024-06-12:
i can't get it to change :/

ved-rivos, 2024-06-12:
Maybe it will respond to a "font-size" tag in ulist?

ulist:
  marker:
    disc:
      content: "\u26ab"
    circle:
      content: "\u26ab"
    square:
      content: "\u25a0"
    font-family: JetBrainsMono
    font-size: ????

wmat, 2024-06-12:
that's exactly what I tried.

ved-rivos, 2024-06-12:
Maybe [U+2022](https://www.compart.com/en/unicode/U+2022) is better?

ved-rivos, 2024-06-18:
I tried a bunch of these and seems like what you have currently is  the best among all.

closed 2024-06-18 by ved-rivos