riscv-non-isa/riscv-iommu #351: When an IOMMU interrupt occurs, is icvec configured by hardware or software?

18772820305, 2024-06-12:
Interrupt-cause-to-vector register maps a cause to a vector. All causes can be mapped to the same
vector or a cause can be given a unique vector. When an IOMMU interrupt occurs, is icvec configured by hardware or software?

ved-rivos, 2024-06-12:
Usually in most devices, configurations are provided by software. Hardware may have defaults for certain configurations. In case of the RISC-V IOMMU, reset defaults are specified for a small set of configuration as listed in section 5.2. For the register fields not explicitly listed in section 5.2, the reset default is UNSPECIFIED. The section 6.2 of the specification includes guidelines on initializing the IOMMU registers. 

Please see the following sections to understand how the `icvec` is used:

- Section 5.27 specifies the `icvec` register to configure interrupt cause to interrupt vector. 
- Section 6.2 - steps 9 through 11 - specify the software guidelines to initialize the `icvec` register. 
- Section 5.18 specifies how the configurations in `icvec` are used by the hardware to select the wire to drive when wire-signaled-interrupts is used (`fctl.WSI=1`) and to select the address/data pair for message-signaled-interrupts when MSI is used (`fctl.WSI=0`). 

Issue #299 has some additional information that may be helpful in understanding how the `icvec` works.

ved-rivos, 2024-06-13:
Hope that helped. Please feel free to ask if there are further questions.

closed 2024-06-13 by ved-rivos

yanhe234, 2024-10-10:
So my understanding here is that software configures icvec, and then hardware reads the icvec register and selects msi_table for subsequent operations, is that correct? sir

ved-rivos, 2024-10-10:
Yes. Thats correct.