riscv-non-isa/riscv-iommu #356: How is the requester ID assigned in an invalid request TLP packet?

dongxieDD, 2024-06-21:
In the invalid TLP packet, the device ID and requester ID are sent to the RC by the IOMMU. The device ID can be determined by the RID, so how should the requester ID be assigned?
[image]
[image]

ved-rivos, 2024-06-21:
It's the host's requester ID.

dongxieDD, 2024-06-24:
Thank you very much for your reply.

18772820305, 2024-06-24:
How to understand the host's request ID?

ved-rivos, 2024-06-24:
It is usually the requester ID of the PCIe root port to which the the device is connected either directly or though switches. The configuration of the PCIe root complex is typically done by system firmware and OS that executes on the host as part of boot. This includes assigning the bus number for the PCIe root ports and the bus numbers for the secondary and subordinate buses. See example of PCI-PCI bridge configuration [here](http://www.science.unitn.it/~fiorella/guidelinux/tlk/node76.html#SECTION00862100000000000000) and [here](https://codywu2010.wordpress.com/2015/11/29/how-modern-multi-processor-multi-root-complex-system-assigns-pci-bus-number/).

closed 2024-06-24 by ved-rivos