riscv-non-isa/riscv-iommu #360: Question about iotval2

hankin996, 2024-06-26:
I have a question about the configuration of bit 1 of iotval2 in FQ data structure.
Based on my understanding of the underlined content in the screenshot below, the bit 1 of iotval2 set 1when the guest page fault is generated by implicit access to vs-stage page table and the result of the access indicates that iommu need automatically updates the A/D of the vs stage page table. 
But now there is a question, following the two-stage translation process, when a guest page fault occurs, there is no implicit access to the next step of vs stage, and it is unclear whether atomatically update to the A/D of the vs stage page table are needed.
Could you please provide a specific description of when bit 1 of iotval2 is set.
Thanks！
[image]

ved-rivos, 2024-06-26:
With two stage page tables, all memory accesses are subject to address translation by the VS-stage and G-stage. To perform this address translation there is a need to load entries from the VS-stage page tables. These implicit load accesses to the VS-stage page table entries are also subject to G-stage page table translation and may cause a guest page fault - in which case the iotval2 bit 1 is 0 but iotval2 bit 0 is 1. For the leaf PTE in VS-stage there may be  a need to set the A and/or D bit. This implicit access to set the A and/or D bit in the VS-stage leaf PTE is also subject to G-stage page table translation and may cause a guest page fault - in which case the iotval2 bit 1 is 1 and iotval2 bit 0 is also 1. The bit 0 of the iotval2 indicates if the guest page fault was a result of an implicit access to the VS-stage page table entry. The bit 1 indicates if that implicit access was a load or a store. As the cause code for guest page fault always corresponds to the type of the original access i.e. a load guest page fault, store/AMO guest page fault, or an instruction guest page fault - the iotval2 helps provide more information when it was the implicit access to the VS-stage PTEs caused by the original access that lead to the fault.

ved-rivos, 2024-06-28:
I hope that addressed your question. Please feel free to ask if there are further questions.

closed 2024-06-28 by ved-rivos