riscv-non-isa/riscv-iommu #366: what should interrupt do when records in fault queue ?

baimengwei, 2024-06-28:
let wire interrupt as an example. When two faults occur in fault queue in time series, and the first fault record is stored in fault queue, the interrupt is assert, then the software detect the interrupt and start to deal with the fault record and deassert the fip. However, at the same time, the next fault record is arrive and stored in fault queue but software not detect it and the interrupt is deassert, software lost a attention which the next fault record is stored in fault queue?

is the software do right ? or the interrupt stay assert when records in fault queue rather than this is a new record ?

ved-rivos, 2024-06-28:
The guidelines for handling interrupts are provided in section 6.5 of the specification. The sequence to follow is to first clear `fip` and then read the the `fqt`/`fqh` to process the fault records. In your example, if this sequence is followed then the `fqt` and `fqh` will be 1 record apart leading to software processing one fault record. A second fault record if produced after clearing of `fip` will cause `fip` to be set again but software may not observe it. This is however okay as when the interrupt service routine completes there is another interrupt pending and the ISR will be invoked again to handle the second interrupt.

closed 2024-07-01 by baimengwei

wangyongzhen0322, 2024-07-01:
I don't quite understand this sentence.
> A second fault record if produced after clearing of fip will cause fip to be set again but software may not observe it.  This is however okay as when the interrupt service routine completes there is another interrupt pending and the ISR will be invoked again to handle the second interrupt.

So if the bit fip is set to 0 from 1 because of software clearing and the hardware wants to set the fip to 1 again because of a fault recorded produced in the fq **at the same cycle**, should the fip set to 0 from 1 at the first cycle and set to 1 again at the second cycle? Or it is not necessary to set the fip to 1 again. It is just okey to wait another fault record to set the fip to 1 and both of fault records will resolved together.