riscv-non-isa/riscv-iommu #371: Question about handling different interrupt pending source simultaneously

viktoryou, 2024-07-02:
For the interrupt source cip, fip, pip, pmip, it seems that they are mutually independent and allowed to be pending simultaneously.

From the guideline in the chapter 6.5 of spec, the flow of handling individual pending source is clear. However, I could not find an explanation of the mixed case of different interrupt sources. For example, at the time or after that cip switched from 0 to 1, what is expected when the fip also switched from 0 to 1, if possible? Also, would the handling of WSI or MSI mode be identical in such case?

ved-rivos, 2024-07-02:
The interrupt sources may assert interrupts independently and concurrently. The interrupt service routines for these may also execute concurrently on different harts in the system. When multiple interrupts target the same hart, which interrupt has priority and whether the ISR of one interrupt can be preempted by another interrupt is software policy but is no different than handling interrupts from any other source. For handling of each interrupt source, the guideline in section 6.5 applies.

viktoryou, 2024-07-02:
From my understanding, in the case where different interrupt sources are asserted concurrently, an implementation of IOMMU should be capable of generating all relative interrupts, instead of picking either one of them and ignoring the other. Is that right?

ved-rivos, 2024-07-02:
Yes, there are 4 independent interrupts sources in the IOMMU - command queue, fault queue, page-request queue, and performance monitoring unit. They should be able to assert their interrupts concurrently. Of course generating an MSI implies doing a memory write and when multiple of these sources want to generate a memory write  for MSI they need to arbitrate  for the bus. There is no need to prioritize in any specific way for this arbitration - these writes for generating an MSI may be delayed a few cycles due to this arbitration.

closed 2024-07-02 by viktoryou