riscv-non-isa/riscv-iommu #379: How to understand the unsupport in c model about command illegal ?

baimengwei, 2024-07-16:
https://github.com/riscv-non-isa/riscv-iommu/blob/main/src/iommu_in_memory_queues.adoc#command-queue-cq

> A command is unsupported if it is defined but not implemented as determined by the `IOMMU capabilities register`.

In the c model, if a iofence.c command is send, but the `g_reg_file.fctl.wsi == 0 && command.iofence.wsi == 1`  is meet the condition, the command will be illegal. However, it is not  involved in a judgement about `IOMMU capabilities register`. 
```
        case IOFENCE:
            if ( command.iofence.reserved != 0 || command.iofence.reserved1 != 0 )
                goto command_illegal;
            // The wired-signaled-interrupt (WSI) bit when set to 1
            // causes a wired-interrupt from the command
            // queue to be generated on completion of IOFENCE.C. This
            // bit is reserved if the IOMMU supports MSI.
            if ( g_reg_file.fctl.wsi == 0 && command.iofence.wsi == 1)
                goto command_illegal;
```
https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_command_queue.c#L160

If a iodir command is send, but command illegal judgement is not involved in ddtp.iommu_mode， which might cause a unexpect condition.  This is also not  involved in a judgement about `IOMMU capabilities register`. And c model use a g_max_devid_mask to instead.
```
            switch ( command.any.func3 ) {
                case INVAL_DDT:
                    // The PID operand is reserved for the
                    // IODIR.INVAL_DDT command.
                    if ( command.iodir.pid != 0 ) goto command_illegal;
                    // When DV operand is 1, the value of the DID operand must not
                    // be wider than that supported by the ddtp.iommu_mode.
                    if ( command.iodir.dv &&
                         (command.iodir.did & ~g_max_devid_mask) ) {
                        goto command_illegal;
                    }
```

https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_command_queue.c#L114

So.  Is the ftcl should not involved in judgement? Or is the ddtp.iommu_mode should involved in judgement?

Thanks.

ved-rivos, 2024-07-16:
> A command is unsupported if it is defined but not implemented as determined by the IOMMU capabilities register.

This statement is about the command itself. The first statement of that paragraph applies to commands that are implemented:  "A command is determined to be illegal if it uses a reserved encoding or if a reserved bit is set to 1."

baimengwei, 2024-07-17:
Get it. Thanks a lot.

closed 2024-07-17 by baimengwei