riscv-non-isa/riscv-iommu #380: Should a set of OF in iohpmctr1-31 CSR generate a pmip

viktoryou, 2024-07-17:
From the spec,
> The OF bit is set when the iohpmcycles counter overflows, and remains set until cleared by software. 

From my understanding, It is also allowed to set the OF by software since the bit has the attribute of RW instead of RW1S. Would the behavior that setting the OF by software be regarded as another counter overflows?

> If the iohpmcycles counter overflows when the OF bit is zero, then a HPM Counter Overflow interrupt is generated by setting ipsr.pmip bit to 1. If the OF bit is already one, then no interrupt request is generated. Consequently the OF bit also functions as a count overflow interrupt disable for the iohpmcycles.

Then, should a HPM Counter Overflow interrupt be generated due to setting the OF by software from zero?

ved-rivos, 2024-07-17:
It is only the "counter overflows when OF bit is zero" that can cause the interrupt. It is not a 0->1 edge of the OF that sets the interrupt pending. It is the carry out from the counter when OF is 0 that sets the interrupt pending.

closed 2024-07-17 by viktoryou

pkennedyTT, 2025-11-12:
> It is only the "counter overflows when OF bit is zero" that can cause the interrupt. It is not a 0->1 edge of the OF that sets the interrupt pending. It is the carry out from the counter when OF is 0 that sets the interrupt pending.

This makes sense to me as there would otherwise be no way to disable counter overflow interrupts without inducing one. But in the description for the `pmip` field in 6.18, the spec says:

> The performance-monitoring-interrupt-pending is set to 1 when OF bit in iohpmcycles or in any of the iohpmctr1-31 registers transitions from 0 to 1.

I interpreted this as meaning regardless of whether that transition was caused by an actual overflow or by a write from software. I assume this should be "...transitions from 0 to 1 due to a counter overflow".