riscv-non-isa/riscv-iommu #382: question about READ_FOR_EXECUTE_TRANSACTION

wanwan0409, 2024-07-23:
in the model, 
when a UNTRANSLATED req perform a READ & EXECUTE operation，and report a fault，pid_valid will be used as a criterion. the result is TTYP=UNTRANSLATED_READ_FOR_EXECUTE_TRANSACTION.
For a TRANSLATED req doing the same,  pid_valid will not be used as a condition for judging the TTYP=TRANSLATED_READ_FOR_EXECUTE_TRANSACTION.

in the spec, as shown in Fig，there is no mention of the need to distinguish between UNTRANSLATED req and TRANSLATED req.
which one is correct？

[image]

ved-rivos, 2024-07-23:
Address translation services (ATS) is a PCIe capability and Translated/Translation requests are generated when ATS is in use. In PCIe, execute can be requested only through the use of PASID prefix. In the spec text you quoted - see the text in parenthesis - e.g, using a PCIe PASID.

wanwan0409, 2024-07-24:
I understand. Thanks a lot.

closed 2024-07-24 by wanwan0409

yanhe234, 2024-10-28:
Hello sir, currently no PCIe Message Request has been found in the reference model, while the other 8 types have definitions, such as PCIe ATS Translation Request. What is the reason for this? Looking forward to your reply very much

ved-rivos, 2024-10-28:
Please see ats_msg_t in iommu_ats.h

yanhe234, 2024-10-29:
Sorry sir, let me rephrase it. Regarding the definition of TTYP in iommu_translate. c, I did not find a PCIe Message Request. However, I did find a PCIe ATS Translation Request. Looking forward to your reply

ved-rivos, 2024-10-29:
> Sorry sir, let me rephrase it. Regarding the definition of TTYP in iommu_translate. c, I did not find a PCIe Message Request. However, I did find a PCIe ATS Translation Request. Looking forward to your reply

Please see:
```
// The TTYP field reports inbound transaction type
// Fault record `TTYP` field encodings
// |TTYP   | Description
// |0      | None. Fault not caused by an inbound transaction.
// |1      | Untranslated read for execute transaction
// |2      | Untranslated read transaction
// |3      | Untranslated write/AMO transaction
// |4      | Reserved
// |5      | Translated read for execute transaction
// |6      | Translated read transaction
// |7      | Translated write/AMO transaction
// |8      | PCIe ATS Translation Request
// |9      | Message Request
// |10 - 31| Reserved
// |31 - 63| Reserved for custom use
#define TTYPE_NONE                                0
#define UNTRANSLATED_READ_FOR_EXECUTE_TRANSACTION 1
#define UNTRANSLATED_READ_TRANSACTION             2
#define UNTRANSLATED_WRITE_AMO_TRANSACTION        3
#define TRANSLATED_READ_FOR_EXECUTE_TRANSACTION   5
#define TRANSLATED_READ_TRANSACTION               6
#define TRANSLATED_WRITE_AMO_TRANSACTION          7
#define PCIE_ATS_TRANSLATION_REQUEST              8
#define PCIE_MESSAGE_REQUEST                      9

```

yanhe234, 2024-10-29:
Yes, I saw the location you indicated, but under what circumstances would TTYP be assigned as PCIe Message Request

` TTYP = TTYPE_NONE;
    if ( req->tr.at == ADDR_TYPE_UNTRANSLATED && req->tr.read_writeAMO == READ ) {
        if ( req->exec_req )
            TTYP = UNTRANSLATED_READ_FOR_EXECUTE_TRANSACTION;
        else
            TTYP = UNTRANSLATED_READ_TRANSACTION;
    }
    if ( req->tr.at == ADDR_TYPE_UNTRANSLATED && req->tr.read_writeAMO == WRITE )
        TTYP = UNTRANSLATED_WRITE_AMO_TRANSACTION;
    if ( req->tr.at == ADDR_TYPE_TRANSLATED && req->tr.read_writeAMO == READ ) {
        if ( req->pid_valid && req->exec_req )
            TTYP = TRANSLATED_READ_FOR_EXECUTE_TRANSACTION;
        else
            TTYP = TRANSLATED_READ_TRANSACTION;
    }
    if ( req->tr.at == ADDR_TYPE_TRANSLATED && req->tr.read_writeAMO == WRITE )
        TTYP = TRANSLATED_WRITE_AMO_TRANSACTION;

    if ( req->tr.at == ADDR_TYPE_PCIE_ATS_TRANSLATION_REQUEST )
        TTYP = PCIE_ATS_TRANSLATION_REQUEST;

    // Has to be one of the valid ttypes - this only for debug - not architectural
    if ( TTYP == TTYPE_NONE ) *((char *)0) = 0;
`

ved-rivos, 2024-10-29:
> Yes, I saw the location you indicated, but under what circumstances would TTYP be assigned as PCIe Message Request

It will be assigned as PCIe message request when it is a PCIe message transaction. PCIe messages that the IOMMU handles are not routed by address and there is no address translation needed for them. So you would not find any instances of message handling in iommu_translate.c

The two messages that the IOMMU receives are - Page Request and Invalidation Completion. For handling of Page Request see `handle_page_request` and for handling of invalidation completion see `handle_invalidation_completion`.

yanhe234, 2024-10-29:
Sir, how do you understand PCIe message transactions? I understand that when connecting PCIe devices, the access initiated should belong to PCIe message transactions, right? Can you provide more details about this issue? thank you
> It will be assigned as PCIe message request when it is a PCIe message transaction.
> 
> PCIe messages that the IOMMU handles are not routed by address and there is no address translation needed for them. So you would not find any instances of message handling in iommu_translate.c

yanhe234, 2024-10-30:
many  thanks