riscv-non-isa/riscv-iommu #383: Question about the transaction type constraint in the model

viktoryou, 2024-07-24:
In the [code](https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_translate.c#L47) , the model judges the execution transaction type as the following called function `get_attribute_from_req` does. This confused me a little. Should IOMMU hardware or the DMA master be responsible to guarantee that it is signaling the correct type of transaction (e.g. execution type)？

If the DMA master is required to do so, then could I say that these constraints of transaction type only be applied to the debug request that independently generated by software?

ved-rivos, 2024-07-24:
> If the DMA master is required to do so, then could I say that these constraints of transaction type inside IOMMU only be applied to the debug request that independently generated by software?

It is not the DMA master responsibility check for validity of the transaction. The checking for validity of the transaction lies on the root complex - which is the collection of root port, host bridge, and the IOMMU.

viktoryou, 2024-07-25:
> The checking for validity of the transaction lies on the root complex - which is the collection of root port, host bridge, and the IOMMU.

Specifically, would IOMMU itself need to check the validity of the transaction, like pv=0, IOMMU ensures pid=0, priv=0 and similar constriant on execution type?

ved-rivos, 2024-07-25:
That is implementation specific and is why `get_attribs_from_req` is provided by the test application.

closed 2024-07-25 by viktoryou