riscv-non-isa/riscv-iommu #384: Question about is_msi flag return of a translated transaction with t2gpa clear

viktoryou, 2024-07-24:
From the spec, in chapter 2.6
> When a MSI PTE is configured in MRIF mode, a MSI write with data value D requires the IOMMU to set the interrupt-pending bit for interrupt identity D in the MRIF. A translation request from a device to a GPA that is mapped through a MRIF mode MSI PTE is not eligible to receive a translated address. This is accomplished by setting "Untranslated Access Only" (U) field of the returned response to 1.

It seems that a translation request from a device to a GPA that is mapped through a basic mode MSI PTE is not mentioned here and could be possibly allowed.

Also, when request is a translated request and `DC.tc.T2GPA` is 0 then the translation process is complete. Under this circumstance, the model would always return the is_msi flag as 0, see [code](https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_translate.c#L177). From my understanding, this would be an issue. Although [here](https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_translate.c#L365) has caches the is_msi flag, the request still unable to get the is_msi flag since translation process completes after DDW. Once the translated request with t2gpa clear finished the access, the is_msi flag may return wrong. Could you help to clarify that?

viktoryou, 2024-07-25:
In such a case, a request accessed IOMMU by ATS protocol and later its GPA was found as a MSI address. After translation it doesn't returns is_msi flag to the device. Again the request issued and hitted devATC. Then it accessed IOMMU as a translated transaction and lost the is_msi infomation and acted as a ordinary transaction.
Could this case happen? Would it become a potential problem?

ved-rivos, 2024-07-25:
The is_msi is never returned to device in ATS translation responses. No such field exists in PCIe protocol.
So devATC never has to worry about is_msi.

The following cases occur for ATS translation requests:
```
is_msi   is_mrif  T2GPA
0            0           0              SPA returned to devATC
0            0           1              GPA returned to devATC. Translated requests
                                           carry GPA.
0            1           X              Cannot happen
1            0           0              SPA returned to devATC. Translated requests carry
                                           SPA - no action needed.
1            0           1              GPA returned to devATC. Translated request carry GPA 
                                           and will be translated by the MSI PT in basic mode.
                                           This translation for the translated request is returned to
                                           the host bridge - not to devATC.
1            1           X              Untranslated requests only response returned
```

ved-rivos, 2024-07-27:
That scenarios is correct. The translated request holds a SPA and the memory type indicated to the host bridge by the IOMMU is PMA. If the translated request carried an override such as using the additional memory attributes in the request then the host bridge will apply the override otherwise the memory request gets the physical memory attribute (PMA) corresponding to the SPA.

viktoryou, 2024-07-29:
> If the translated request carried an override such as using the additional memory attributes in the request then the host bridge will apply the override otherwise the memory request gets the physical memory attribute (PMA) corresponding to the SPA.

Currrently the model would always return PMA as the value of PBMT and the host bridge would fail to do so, see the [code](https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_translate.c#L175). I am wondering if it is eligible to receive a translated address when the memory attributes is overridden. Could you help to clarify that?

ved-rivos, 2024-07-29:
For a translated request the IOMMU should always return the memory attribute as PMA. Whether the host bridge performs any overrides is implementation specific.

closed 2024-07-29 by viktoryou