riscv-non-isa/riscv-iommu #385: Question about gpasize in iotval2 when s1 bare and gpa over 56bit

Orion3-e, 2024-07-25:
According to the spec 
[image]
seems the iotval2 should save the whole gpa high part value [63:2],  even the gpa is over maxgpa , 
but in the c model , seems gpa dont save the value above 56bit when  S1 bare (va=gpa),  depending on page_sz .
[image]
[image]
so , should the iotval2 save the upper_bits in this situation? thanks

ved-rivos, 2024-07-25:
Do you have a test input that you can add to show what you mean? What was your input va and what do you see saved in iotval2 for that input?

Orion3-e, 2024-07-26:
[image]
this is the req we send : iova = fffffff54d1ee2bc 
[image]
this is the result covert from the rivos model 
[image]
this is the two stage function handles gpa for this req , the page_sz screenshoted in my last comment shows it is not 64 , causing the gpa high part [63:57] discarded
in this situation , dc.pdtv=1 && vsatp mode=bare so s1 bare va =gpa , so i think the iotval2 should save fffffff54d1ee2bc ,not the fffff54d1ee2bc in the temp fault q result rivos , seems bit [63:57] of iotval2 is discarded by two stage translation. Thanks

ved-rivos, 2024-07-27:
The issue is that the pte->PPN is 44 bit wide. I have fixed this in #390 
Please also send me a direct message.

Orion3-e, 2024-07-29:
Now the iotval2 check of that req pass,  thanks

closed 2024-07-29 by ved-rivos

zetalog, 2024-07-30:
I have different opinion of the saved iotval/iotval2, probably you guys should take a look over it:
https://github.com/riscv-non-isa/riscv-iommu/issues/391