riscv-non-isa/riscv-iommu #391: Missing DMA_MASK in IOMMU specification and input addresses are used in the wrong way

zetalog, 2024-07-30:
In the modern operating systems, DMA addresses are not just physical addresses or virtual addresses, please refer to:
https://github.com/torvalds/linux/blob/master/Documentation/core-api/dma-api-howto.rst
Operating systems treat DMA addresses as "bus addresses", which are likely referring to a subset of a physical address space or a subset of a virtual address space.
In the systems like Linux, there is a DMA_MASK attributes system widely applied or device specific.
When a virtual/physical address is to be converted into a DMA address, operating system simply masks off the the most significant bits of the DMA address, that means, the most significant bits of the DMA address are always 0s.

And there are many ecosystem DMA master IPs which support limited address width, say, 16-bits, 32-bits, or 48-bits. Such IPs do not know whether the programmed DMA addresses are virtual addresses or physical addresses. The integration guide of such silicon won't introduce a design forcing such IPs to be integrated with most significant bits wired to 1s (ex, same as bit 48).

However in IOMMU specification, it requires fault to be generated for virtual addresses whose most significant bits must follow virtual address requirement (e.x., 63-48 should be same as 48 for Sv48, etc.). When an input address is reported as fault, iotval/iotval2 in the fault queue should be filled in with full 64-bit input address, while actually, software and hardware only capable of recording masked DMA bus addresses.

This issue cause serious application problems in the operating system. When an IOMMU implementation is implemented to be compliant to the specification, requiring fault to be reported for non-compliant virtual addresses whose most significant bits are not equal to bit 48, wrong fault can be seen in Linux preventing DMA remapped DMA transfers to be submitted from such DMA masters whose DMA_MASK is not 64-bit and wires most significant bits of the output DMA addresses to 0s. A stupid workaround could be to wire those bits to bit 48. Then another compliant problem can be seen in the IOMMU fault queue, whose report may contain a record whose iotval/iotval2 should be physical address but with most significant bits wired to 1s.

IMHO, IOMMU specification should introduce DMA_MASK capability indicating supported input address of the IOMMU device,  and a similar DMA_MASK field in the device context indicating supported DMA address width of the DMA master device. Most significant bits beyond DMA_MASK are allowed to be all zeros (probably should all zeros or all ones to allow design variations). And the virtual address sanity check should be bounded into the DMA_MASK range, for the address bits beyond the DMA_MASK, all checks should be ignored or we can just apply a simple check validating if all such bits equal to bit[DMA_MASK] like all other DMA remapping silicon design do.

ved-rivos, 2024-07-30:
The IOMMU provides the address that was submitted for address translation in the fault record. Software may apply masks as appropriate to the reported addresses .

zetalog, 2024-07-30:
> The addresses presented to the IOMMU are addressses generated by the IP. If the IP cannot generate an address beyond say 16-bits then the IOMMU would not see any addresses requested for translation bits beyond what the IP can generate. There is no requirement to wire most significant bits to all 1s etc.

Then we may introduce a configurable "DMA_MASK" option into the reference model, ignoring significant bits checking if it resides beyond the configured DMA_MASK.

ved-rivos, 2024-07-30:
> Then we may introduce a configurable "DMA_MASK" option into the reference model, ignoring significant bits checking if it resides beyond the configured DMA_MASK.

That would be incorrect thing to do. The IOMMU should prevent misbehaving, misprogrammed and/or malicious devices from accessing memory that they are not authorized to access. Masking the address will in best case hide the misprogramming and in worst case hide the malicious behavior or misbehavior.

zetalog, 2024-07-30:
The following code introduces wrong fault in Linux, preventing such DMA masters to be used:
```
    // 1. Let a be satp.ppn × PAGESIZE, and let i = LEVELS − 1. PAGESIZE is 2^12. (For Sv32,
    //    LEVELS=2, For Sv39 LEVELS=3, For Sv48 LEVELS=4, For Sv57 LEVELS=5.) The satp register
    //    must be active, i.e., the effective privilege mode must be S-mode or U-mode.
    if ( iosatp.MODE == IOSATP_Sv32 && SXL == 1 ) {
       ...
        // When `SXL` is 1, the following rules apply:
        // * If the S/VS-stage page table is not `Bare` then a page fault corresponding to
        //   the original access type occurs if the `IOVA` has bits set beyond bit 31.
        mask = (1UL << (64 - 32)) - 1;
        masked_upper_bits = (iova >> 32) & mask;
    }
    if ( iosatp.MODE == IOSATP_Sv39 && SXL == 0 ) {
        ...
        mask = (1UL << (64 - 38)) - 1;
        masked_upper_bits = (iova >> 38) & mask;
    }
    if ( iosatp.MODE == IOSATP_Sv48 && SXL == 0 ) {
        ...
        mask = (1UL << (64 - 47)) - 1;
        masked_upper_bits = (iova >> 47) & mask;
    }
    if ( iosatp.MODE == IOSATP_Sv57 && SXL == 0 ) {
        ...
        mask = (1UL << (64 - 56)) - 1;
        masked_upper_bits = (iova >> 56) & mask;
    }
    // Instruction fetch addresses and load and store effective addresses,
    // which are 64 bits, must have bits 63:<VASIZE> all equal to bit
    // (VASIZE-1), or else a page-fault exception will occur - for SXL=0
    // Do the address is canonical check - for SXL=0
    // For SXL = 1 check bits 63:32 are all 0
    if ( (masked_upper_bits != 0 && masked_upper_bits != mask && SXL == 0) ||
         (masked_upper_bits != 0 && SXL == 1) ) goto page_fault;
```

IMO, we should alter this piece a code a bit to allow an SoC specific DMA_MASK to be configured to prevent wrong fault to be generated for the DMA masters whose address width is less than 39, or less than 48.

This is useful when a verification TB is created with IOMMU + DMAC (address width is less than 39/48) + reference model.

ved-rivos, 2024-07-30:
> IMO, we should alter this piece a code a bit to allow an SoC specific DMA_MASK to be configured to prevent wrong fault to be generated for the DMA masters whose address width is less than 39, or less than 48.

Consider that a DMA master can generate only generate addresses less than 48 bit, say 16-bit wide addresses. Such a DMA master can only generate addresses 0x000000000000 through 0x00000000FFFF legally. When Sv48 is in use, all of these addresses are legal addresses. The range of legal addresses for Sv48 are 0x0000'0000'0000'0000 to 0x0000'7FFF'FFFF'FFFF and 0xFFFF'8000'0000'0000 to 0xFFFF'FFFF'FFFF'FFFF. Could you please give an example of such an address?

zetalog, 2024-07-30:
Real issue is seen against a DMAC in an SoC design, whose master can only support maximum 44-bits, DMAC designer then wires higher bits of the DMA addresses to 0s.

When we use such DMAC in Linux kernel with IOMMU remapping enabled, following kernel panic can be seen due to IOMMU VA checking logic, preventing a valid DMA transfer to be performed:

[image]

zetalog, 2024-07-30:
OK, I see you mean 0x7fff_ffff_x000 shouldn't be a valid address for such kind of DMA device due to its driver doesn't invoke dma_set_mask(44). So you mean there is no worry about seeing an input address between 0xFFFF'8000'0000'0000 and 0xFFFF'FFFF'FFFF'FFFF. And when such an test bench is created, input address stimuli shouldn't controlled in the TB side rather than in the IOMMU reference model.

zetalog, 2024-07-30:
One more concern is related to the IOMMU device capability. What if the IOMMU only accepts 44-bits input address, and the DMA masters are configured to use maximum 44-bits input address as well.
In such a case, shall we introduce such kind of capability in the IOMMU reference model?

zetalog, 2024-07-31:
OK, this makes root cause clearer.
Thanks and best regards.

zetalog, 2024-12-06:
For PAS field, since it reflects the bit width of DMA master or bus capability the IOMMU is connected to.
For DMA masters and bus interconnects, higher bits will always be truncated to 0s, the padding rule is not compatible with RISC-V virtual addresses.
Thus PAS actually reflects the bit width of both the input address (could be virtual) and the output address, it actually is not "physical".
Shall we use bus address size of dma address size instead of PAS?

ved-rivos, 2024-12-06:
> For PAS field, since it reflects the bit width of the DMA masters or the bus interconnects which the IOMMU is connected to. 

The PAS field reflects the physical address space addressable by the IOMMU and not the bit width of DMA masters.

zetalog, 2024-12-09:
Not sure if we are synchronized on the same page. Let me describe the scenarios I'm considering of.

Considering an Soc system, DMA is capable of addressing 40-bit DMA addresses. After being translated by the IOMMU, the 40-bit is translated into 44-bit bus addresses to be connected to the bus interconnects.
[image]
In OSen, when the DMA master is being programmed using a not-bare DMA remapping facility (IOMMU), the 40-bit dma_mask will be applied to the virtual addresses.

The application problem is:
Considering a scenario, that SoC designs want to combine multiple DMA masters into a sub IO networking with bus interconnects to save the number of the required SIDs:
[image]

ved-rivos, 2024-12-09:
> Considering an Soc system, DMA is capable of addressing 40-bit DMA addresses. After being translated by the IOMMU, the 40-bit is translated into 44-bit bus addresses to be connected to the bus interconnects (PAS is 44).

The PAS is not related to any of this. The PAS in the specification is what the IOMMU can address.

> In the first scenario, how could IOMMU to limit the width of the input addresses? I could only see PAS in the IOMMU specification.

The driver of DMAC1 should only program an address below 2^40 and the driver of DMAC0 an address below 2^44. 

 > For the second scenario, when the DMA masters issue the DMA requests with the VAs whose higher bits (bit 38 or bit 47 is one concerning with Sv39 or Sv48) could be 1, how could the VA bus interconnects restore the higher bits if the DMAC0 / DMAC1 truncate the DMA addresses out?

If the bus of a DMA master is 40 bit wide then it should not be programmed with an address that is wider than 40 bits. Such device will likely not even have registers to hold an address wider than 40 bits.

> Since there is no input address width configurable in IOMMU, should the VA bus restores the address back to 64-bit (bit 63-44 for DMAC0 and bit 63-40 for DMAC1)? Or if we only need to restore the VA address back to 44-bit, how the VA bus interconnect pads the higher bit 40-43 for the DMAC1?

Such buses and interconnects are outside the scope of the IOMMU to specify.

zetalog, 2024-12-09:
> The driver of DMAC1 should only program an address below 2^40 and the driver of DMAC0 an address below 2^44.

For Sv39, bit-38 = 1 is a valid address as 38 is smaller than 40 or 44, and OS kernel can have an DMA address remapped to bit-38 = 1 region. And bit 39 - cpu bus width (ex, 39 - 63) could be padded with bit-38.
When this address is being programmed to the DMA master, the higher bits (bit 40~63 for DMAC1 and bit 44-63 for DMAC1) could be truncated by the DMA controller.
Then the IOMMU will get an address with bit 38-39 is 1 and 40-63 padded with 0 for DMAC1, an address with bit 38-43 is 1 and 44-63 padded with 0 for DMAC0. From IOMMU's point of view, is such a VA a valid Sv39 address?

ved-rivos, 2024-12-09:
> Then the IOMMU will get an address with bit 38-39 padded with 1 by OS and 40-63 truncated to 0 by DMAC1, an address with bit 38-43 padded with 1 by OS and 44-63 truncated to 0 by DMAC0. From IOMMU's point of view, is such kind of VA a valid Sv39 address?

When mode is configured as Sv39/Sv48/Sv57, the VA must be a 64-bit address. There are only two VA lengths - 64 and 32 - supported by the IOMMU. The addresses where bits 63:39 do not match bit 38 are not valid virtual addresses.

Since the devices in this example are not 64-bit capable, they cannot be programmed with a 64-bit VA. They may be programmed with a PA or a GPA. If a VA needs to be used then the OS has to use Sv48 or Sv57 and map the VA below 2^40/2^44 for such devices.

zetalog, 2024-12-10:
Hi, I drew a bug scenario chart below:
 
[image]

And I'll get back to you after correctly organizing my words. Thanks for the patience.

ved-rivos, 2024-12-10:
For such a device, the DMA width should be limited to 39 bits and not 40 bits. Alternatively, one can use Sv48 mode. 

For a device that trucates addresses, any address negative address such as 0xFFFFFFC000100000 is an invalid. Since these devices cannot preserve the upper bits and instead replace them with 0, only positive addresses can be used with such devices. 

For Sv39, the range of legal addresses is as follows:
[image]

For such devices that replace the upper bits with 0, the range of valid addresses is - 0x0000000000000000 to  0x00007FFFFFFFFFFF. Specifically the address 0x000000C000100000 is a non-canonical and illegal address.

zetalog, 2024-12-11:
> For a device that truncates addresses, any address negative address such as 0xFFFFFFC000100000 is an invalid. Since these devices cannot preserve the upper bits and instead replace them with 0, only positive addresses can be used with such devices.

If we use top bits ignorance in IOMMU, the negative addresses could also be valid for such devices.

ved-rivos, 2024-12-12:
When the IOVA is a VA, the rules for VA translation apply. This includes checking for validity of the VA. Such truncated/invalid VAs will not be accepted by x86 IOMMUs. For ARM IOMMUs such truncations may also lead to the wrong TTB being used for translation i.e. get translated by TTB0 instead of TTB1.

zetalog, 2024-12-17:
> For ARM IOMMUs such truncations may also lead to the wrong TTB being used for translation i.e. get translated by TTB0 instead of TTB1.

ARM STE contains both TTB0 and TTB1. Shouldn't RISC-V IOMMU take care of kernel space VA?

ved-rivos, 2024-12-17:
> ARM STE contains both TTB0 and TTB1. Shouldn't RISC-V IOMMU take care of kernel space VA?

Unlike ARM architecture that allows setting up two pages tables - one which is used when upper address bits are 0 and one which is used when upper address bits are 1 - the RISC-V virtual memory system has a single page table which is used to map both forms of addresses. Whether kernel space VAs are mapped with upper address bits set to all 1 or all 0 is a software convention. The software convention in most 64-bit OSs including linux has been to map kernel VA with upper address bits as all 1 i.e. as negative addresses. As noted in the address map I pasted above a 64-bit address space has two distinct ranges of valid virtual addresses - a positive range and a negative range. Separating them is a range of invalid/non-canonical addresses. There is no limitation in using valid addresses from either the negative or positive address sub-ranges.

zetalog, 2024-12-20:
> There is no limitation in using valid addresses from either the negative or positive address sub-ranges.

When a buffer is allocated in the kernel heap (ex., kmalloc), and later turned into a direct IO buffer passed to the DMA. The DMA could treat this VA as sanitized VA and issue an IO bus RDWR out. While this could make IOMMU to report a failure as BUS might truncate the address, leaving higher bits zeroed. If IOMMU couldn't ignore the higher bits its incoming connector doesn't support, but treating all incoming VA as a whole 64-bit value, how could IOMMU handle this address?
How this VA can be correctly translated by IOMMU without any performance downgrade? IMO, the fastest way is to ignore the higher bits where no one cares them in the entire system.

ved-rivos, 2024-12-20:
A device that truncates addresses is not compatible with shared virtual addressing and the iova cannot be a VA. It can be a GPA or a SPA. In an earlier post there was a question about TTB0 and TTB1. Kernel VA with upper bits set are only mapped in TTB1 and when the device truncates its address the iommu in such systems would incorrectly use TTB0.

zetalog, 2024-12-25:
Let me try to use a table to describe the ambiguity.
There could be 2 solution to approach when DMA address size < 64-bit.
1. Sign extends the highest bits of the DMA address and compares it with the correct virtual address.
2. Truncates the correct virtual address and compares it with the DMA address.
Considering the IOMMU only support Sv39 and Sv48 for simplicity.

For approach 1:
| DMA address size | Translation mode | Negative address | Extended bit | Fault | Descriptions          |
| ---------------- | ---------------- | ---------------- | ------------ | ----- | --------------------- |
| >= 48            | Sv39             | No               | VA[38]       | No    |                       |
| >= 48            | Sv39             | Yes              | VA[38]       | No    |                       |
| >= 48            | Sv48             | No               | VA[47]       | No    |                       |
| >= 48            | Sv48             | Yes              | VA[47]       | No    |                       |
| < 48 && >= 39    | Sv39             | No               | VA[38]       | No    |                       |
| < 48 && >= 39    | Sv39             | Yes              | VA[38]       | No    |                       |
| < 48 && >= 39    | Sv48             | No               | VA[47]       | No    | VA[47] is 0 (valid)   |
| < 48 && >= 39    | Sv48             | Yes              | VA[47]       | Yes   | VA[47] is 0 (invalid) |
| < 39             | Sv39             | No               | VA[38]       | No    | VA[38] is 0 (valid)   |
| < 39             | Sv39             | Yes              | VA[38]       | Yes   | VA[38] is 0 (invalid) |
| < 39             | Sv48             | No               | VA[47]       | Yes   | VA[47] is 0 (invalid) |
| < 39             | Sv48             | Yes              | VA[47]       | Yes   | VA[47] is 0 (invalid) |

For approach 2:

ved-rivos, 2024-12-26:
> As SoC can easily pad an invalid VA into a valid VA (NO. 8/9/10/12/13/14/15/16, please refer to NO. 8/12).

Taking number 8 as an example. The device is only capable of holding 47 bits. This violates the rule:
"The device must be capable of holding and generating n bits if the address translation mode is Sv< n >"

There is no amount of "fixing" that the SoC or the IOMMU can do to make this device compatible with Sv48 as it can only address 2^47 address range out of the 2^48 valid address range.

ved-rivos, 2025-01-09:
Hope that addressed these questions. Feel free to ask if there are further questions.

closed 2025-01-09 by ved-rivos