riscv-non-isa/riscv-iommu #393: Question related to IOMMU CQ command iofence pr/pw options

zetalog, 2024-08-12:
According to the specification, an IOMMU implementation should implement PR/PW options to wait for the IO read/writes to complete.

However in the operating systems, the DMA related APIs are always used in the following sequences:

dma_map -> DMA remap device, i.e., IOMMU
dma_submit_read/write -> DMC channel device, i.e., DMA masters
dma_complete -> DMC channel device, i.e., DMA masters
dma_unmap -> DMA remap device, i.e., IOMMU

There is no possibility that IOFENCE.PR/PW invoked between dma_submit_read/write and dma_complete.
We can also see in the upstream Linux kernel, IOFENCE is only used to ensure TLB flushed before further IOMMU operations. There is no usage model using PR/PW to ensure DMA transfers to complete.

What kind of the motivations and the expected usage models have been discussed by the specification committee when such PR/PW options are presented?

ved-rivos, 2024-08-12:
The informative note in section 3.1.2 discusses the usage of `PR` and `PW` bits at length. I am repeating some of that here:
> Software should ensure that all previous read and writes processed by the IOMMU have been committed to a global ordering point before reclaiming memory that was previously made accessible to a device. A safe sequence for such memory reclamation is to first update the page tables to disallow access to the memory from the device and then use the IOTINVAL.VMA or IOTINVAL.GVMA appropriately to synchronize the IOMMU with the update to the page table. As part of the synchronization if the memory reclaimed was previously made read accessible to the device then request ordering of all previous reads; else if the memory reclaimed was previously made write accessible to the device then request ordering of all previous reads and writes. Ordering previous reads may be required if the reclaimed memory will be used to hold data that must not be made visible to the device,

>We can also see in the upstream Linux kernel, IOFENCE is only used to ensure TLB flushed before further IOMMU operations. 

AFAICT the RISC-V IOMMU patches are not upstream yet. There is a v6 patch set under [review](https://lore.kernel.org/linux-riscv/7a4c5871821a970ac9ee4b77fc1f41abd5b75ed6.1716578450.git.tjeznach@rivosinc.com/T/#m044b7de3c4365b96508c46e68841e0f7945732d9). If there are questions about that patchset then it is best to ask on the lkml. Please also see `riscv_iommu_cmd_iofence()` in that patchset.

zetalog, 2024-08-13:
The question is not related to the RISC-V IOMMU patchset.

The problem is: we couldn't see any useful usage model for PR/PW options.

ved-rivos, 2024-08-13:
> The problem is: we couldn't see any useful usage model for PR/PW options. Could you provide detailed expectations for these options from spec's point of view.

Please see section 3.1.2 for details on usage model for `PR`/`PW`.  After a page is unmapped for DMA, that page may be reassigned to a different application. All DMA to these pages need to be prevented thereafter. This is done by first unmapping the page from the IOMMU page tables and then invalidating the IOMMU address translation cache. These two steps ensure that any new translation requests seen by the IOMMU will get aborted. However there might be DMA requests that have already performed the IOMMU translation but have not yet written to or read from the page. Before the page can be reclaimed for another use such requests also need to be completed. The `PR`/`PW` serves the purpose of ensuring those are completed.

zetalog, 2024-08-14:
> After a page is unmapped for DMA, that page may be reassigned to a different application. All DMA to these pages need to be prevented thereafter.

This is not a valid scenario in the operating system. A page is unmapped for DMA only if all significant DMA transactions to/from this page are completed. Since DMA transactions layer and DMA remapping layer are not the same software layer in the operating system.
Only software bug can create such scenario and I don't think such kind of bug can be seen in the modern OS design.

ved-rivos, 2024-08-14:
I hope you realize that some DMA transactions can be posted. Also one of the reason to have an IOMMU is security against malicious or misbehaving hardware or software - for instance when the hypervisor is un-mapping a page from the guest VM address space - it does not necessarily trust the device or the driver in the VM to do the right thing. BTW, please also see SYNC commands in DTI for both TBU and ATS interface. 

However, all OSs have used such a command for other architectures IOMMUs. But I don't have full insight into your operating system design or IO design so it is possible that the OS/devices you are working with does  not need this functionality. If an OS/platform does not need this functionality then it does not need to use the command.

ved-rivos, 2024-08-19:
Please ask if there are further questions.

yanhe234, 2024-10-29:
Hello sir, how does the software operate when some memory fails due to certain reasons?
My understanding is to invalidate the memory area and wait for the completion of IOMMU invalidation before reallocating this memory area。
Another question: How do you understand the sentence 'First, update the page table to prevent devices from accessing memory'?
> >  A safe sequence for such memory reclamation is to first update the page tables to disallow access to the memory from the device and then use the IOTINVAL.VMA or IOTINVAL.GVMA appropriately to synchronize the IOMMU with the update to the page table.

ved-rivos, 2024-10-29:
> My understanding is to invalidate the memory area and wait for the completion of IOMMU invalidation before reallocating this memory area。

Correct.

> How do you understand the sentence 'First, update the page table to prevent devices from accessing memory'?

If the memory region is mapped into the page table associated with the device then the device can continue to make memory access to that region. So first step is to unmap it from the page table to prevent any further memory accesses from the device.