riscv-non-isa/riscv-iommu #394: The differences in the behavior of the lower 12 bits of iotval2 between the model and the specification.

Taurins-peng, 2024-08-12:
When gxl is 0, sxl is 1, iova is 'hfff_ffff_ffff, iosatp is Bare, and iohgatp is SV39X4, a gpgfault occurs, and the model directly changes the lower 2 bits to 0.
[image]
Expected scenario: According to the specification, iotval2[11:2] can be 0, however, iotval2[1:0] is determined as described below.
[image]
[image]

ved-rivos, 2024-08-12:
Thats right that an IOMMU implementation has the flexibility of a) report the page offset OR b) report 0 for the page offset. An implementation would pick a) or b). The reference implementation does a).

 The bit 0 and 1 indicate if the guest page fault was caused due to an implicit access to translate VS stage page tables. When iosatp is Bare, there are no such implicit accesses i.e. there is no VS-stage page table. In such cases the GPA is just the VA. 

For guest page faults that occur as a result of VS-stage address translation please see [here](https://github.com/riscv-non-isa/riscv-iommu/blob/46fabbd3e37a63071577d1ff0b5551aa12b3b8c2/iommu_ref_model/libiommu/src/iommu_two_stage_trans.c#L380).

ved-rivos, 2024-08-15:
Please ask if there are further questions.

closed 2024-08-15 by ved-rivos