riscv-non-isa/riscv-iommu #395: Does IOMMU need to check the EN_ATS for ATS Invalidation Completion？

wangyongzhen0322, 2024-08-12:
Is it possible that software send a ATS INVAL command whose EN_ATS is 0？ And if it is possible，I think IOMMU needs to check the EN_ATS about the ATS Invalidation Completion.

ved-rivos, 2024-08-12:
The IOMMU processes invalidation completions only if there is an outstanding invalidation being tracked by the IOMMU i.e. software has submitted a invalidation request. Each invalidation request has a invalidation tag (ITAG) and a device may generate as many invalidation completions as the number of Traffic Classes (TCs) it may have used. The IOMMU tracks the pending invalidations using a tracker indexed by ITAG and DID . If a completion is received for which there is no outstanding invalidation then they are discarded as unsupported requests. There is no need to check `EN_ATS` for this action.

wangyongzhen0322, 2024-08-13:
I see. Thanks for your answer.

closed 2024-08-13 by wangyongzhen0322