riscv-non-isa/riscv-iommu #401: About the behavior of clearing cqcsr.cqen.

zhuotianshu, 2024-08-19:
In 5.15. Command-queue CSR (cqcsr):

> When cqen is changed from 1 to 0, the command queue may stay active (with busy asserted) until the commands already fetched from the command-queue are being processed and/or there are outstanding implicit loads from the command-queue.

I'm confused about the "and/or" here. And I also confused that why the condition is "there are outstanding implicit loads" rather than "there are no outstanding implicit loads". Because as I understand it, a queue could be disabled only when entries already been fetched have been complete (or timeout), and no outstanding implicit loads is in-flight.

ved-rivos, 2024-08-19:
The command queue may stay active - with busy asserted - when there are outstanding implicit loads.

zhuotianshu, 2024-08-19:
> The command queue may stay active - with busy asserted - when there are outstanding implicit loads.

I understand that. But it seems that the sentence should be:

When cqen is changed from 1 to 0, the command queue may stay active (with busy asserted) until the commands already fetched from the command-queue are being processed and~~/or~~ there are **no** outstanding implicit loads from the command-queue.

ved-rivos, 2024-08-19:
It stays active until there *are* outstanding implicit loads. When there are no outstanding implicit  loads it becomes inactive.

zhuotianshu, 2024-08-19:
Got that. There may be some grammatical ambiguity for me, because I thought this sentence means "cq may stay active until situation A & situation B are both true". But I can understand that.

closed 2024-08-19 by zhuotianshu