riscv-non-isa/riscv-iommu #402: About the behavior of msi_vec_ctl_x.M.

zhuotianshu, 2024-08-19:
In 5.28. MSI configuration table (msi_cfg_tbl):

> When the mask bit M is 1, the corresponding interrupt vector is masked and the IOMMU is prohibited from sending the associated message. Pending messages for that vector are later generated if the corresponding mask bit is cleared to 0.

What if I continuously toggle the M bit while the corresponding ip bit in ipsr is constantly high? will there be many times of MSI or just once?( A correct software must clear ip bit in ipsr after handling a interrrupt, so I just want to comfirm the hardware behavior in this special case.)

ved-rivos, 2024-08-19:
MSI is edge triggered. When messages are generated if the mask is 1 then those messages will stay pending and get delivered when the mask is cleared. Masking again and unmasking will not lead to further interrupts as there are no more messages pending.

zhuotianshu, 2024-08-19:
> MSI is edge triggered. When messages are generated if the mask is 1 then those messages will stay pending and get delivered when the mask is cleared. Masking again and unmasking will not lead to further interrupts as there are no more messages pending.

Does it mean that in MSI situation, when delivered, ip bit in ipsr will be automatically cleared?

ved-rivos, 2024-08-19:
> Does it mean that in MSI situation, when delivered, ip bit in ipsr will be automatically cleared?

No. A message is generated when a ip bit goes from 0->1 i.e. on the edge.

zhuotianshu, 2024-08-20:
> When messages are generated if the mask is 1 then those messages will stay pending and get delivered when the mask is cleared.

I looked for this behavior in the reference model. I expect that there may be an conditional MSI delivering process when writting `msi_cfg_tbl[x].msi_vec_ctrl.m`, but there seems no such behavior. The reference model code is:

```
write_register(
    uint16_t offset, uint8_t num_bytes, uint64_t data) {
    ...
    switch (offset) {
    ...
        case MSI_VEC_CTRL_0_OFFSET:
        case MSI_VEC_CTRL_1_OFFSET:
        case MSI_VEC_CTRL_2_OFFSET:
        case MSI_VEC_CTRL_3_OFFSET:
        case MSI_VEC_CTRL_4_OFFSET:
        case MSI_VEC_CTRL_5_OFFSET:
        case MSI_VEC_CTRL_6_OFFSET:
        case MSI_VEC_CTRL_7_OFFSET:
        case MSI_VEC_CTRL_8_OFFSET:
        case MSI_VEC_CTRL_9_OFFSET:
        case MSI_VEC_CTRL_10_OFFSET:
        case MSI_VEC_CTRL_11_OFFSET:
        case MSI_VEC_CTRL_12_OFFSET:
        case MSI_VEC_CTRL_13_OFFSET:
        case MSI_VEC_CTRL_14_OFFSET:
        case MSI_VEC_CTRL_15_OFFSET:
            if ( g_reg_file.capabilities.igs == WIS )
                break;
            x = (offset - MSI_ADDR_0_OFFSET) / 16;
            if ( x >= (1UL << g_num_vec_bits) ) 
                break;
            g_reg_file.msi_cfg_tbl[x].msi_vec_ctrl.m = msi_vec_ctrl_temp.m;
            break;
```

It seems that clearing mask bit won't cause an pending MSI to be delivered. And I also didn't find the code pending an MSI (recording the 0->1 edge of ip bits in ipsr when M=1). Have I ignored something?

zhuotianshu, 2024-08-20:
OK, thanks.

closed 2024-08-20 by zhuotianshu