riscv-non-isa/riscv-iommu #404: What type of MSI write access fualt is included in an IOMMU MSI access fault (cause=273)?

zhuotianshu, 2024-08-19:
If a normal MSI from a io-device's memory access request cause an access fualt, will IOMMU report a IOMMU MSI access fault?
If a notice MSI cause an access fualt, will IOMMU report a IOMMU MSI access fault?

ved-rivos, 2024-08-19:
> If a normal MSI from a io-device's memory access request cause an access fualt, will IOMMU report a IOMMU MSI access fault?

The IOMMU does not see memory requests. It only sees address translation requests. If the  memory access for an MSI leads to an access fault then such transactions are aborted by the IO bridge. When a transaction is aborted the response provided to the device depends on the IO protocol. For PCIe write transactions are posted and so there is no response - the write is discarded. For other protocols like AXI a SLVERR may be generated.

> If a notice MSI cause an access fualt, will IOMMU report a IOMMU MSI access fault?

The partitioning of responsibility between IOMMU and IO bridge for recording MSI in an MRIF and generating the notice MSI are implementation specific. An implementation may report a IOMMU MSI access fault if the notice MSI is generated by the IOMMU.

zhuotianshu, 2024-08-19:
That's clear, thanks for replying.