riscv-non-isa/riscv-iommu #405: Question about issue 291.

zhuotianshu, 2024-08-19:
In 2.6. PCIe ATS translation request handling:

> If the translation could be successfully completed but the requested permissions are not present (Execute requested but no execute permission; no-write not requested and no write permission; no read permission) then a Success response is returned with the denied permission (R, W or X) set to 0 and the other permission bits set to the value determined from the page tables.The X permission is granted only if the R permission is also granted.

The conclusion of https://github.com/riscv-non-isa/riscv-iommu/issues/291 is that, the above description should add a restriction on setting the X bit when using a success response to respond an ATS translation request that the requested permissions are not present. 

But in previous paragraph:

> When translation could not be completed due to the following causes a Success Response with R and W bits set to 0 is generated. No faults are logged in the fault queue on these errors. The translated address returned with such completions is UNSPECIFIED.
> • Instruction page fault (cause = 12)
> • Read page fault (cause = 13)
> • Write/AMO page fault (cause = 15)
> • Instruction guest page fault (cause = 20)
> • Read guest-page fault (cause = 21)
> • Write/AMO guest-page fault (cause = 23)
> • PDT entry not valid (cause = 266)
> • MSI PTE not valid (cause = 262)

Since an ATS translation request that the requested permissions are not present will cause a page fault, I thought in this situation, R and W bits will always be set to 0. And since "The X permission is granted only if the R permission is also granted.", X bit will also always be set to 0. So it seems no need to change the spec description.

ved-rivos, 2024-08-19:
> The X permission is granted only if the R permission is also granted.

This is a PCIe rule and not an IOMMU rule. If the device requested say read permission but not execute permission and the translation has both read and execute permission then the permission granted is only read since execute was not requested.

zhuotianshu, 2024-08-19:
> A R=W=0 response occurs when the page table entry is not present...

Yes I know that, but page faults can also occur when permission checks failed. Suppose that an ATS translation request requests read and write permission, but the page only has read permission, then what will the success response look like? Will it be R=1, W=0 or R=W=0? As I understand it, since this request will cause a write page fault (though won't be reported in FQ),  R and W bits in the Success Response should be set to 0 according to:

> When translation could not be completed due to the following causes a Success Response with **R and W bits set to 0** is generated. No faults are logged in the fault queue on these errors. The [...]

So it seems that when permission check fails, the success response is always with R=W=0, and thus X=0?

ved-rivos, 2024-08-19:
The key point is "When translation could not be completed". The specific case you highlighted occurs where the translation could not be completed because it was an invalid address. 
See:
> If the translation could be successfully completed but the requested permissions are not present (Execute requested but no execute permission; no-write not requested and no write permission; no read [...]

zhuotianshu, 2024-08-19:
Got that, thank you very much.

closed 2024-08-19 by zhuotianshu