riscv-non-isa/riscv-iommu #407: Scenario Issues Regarding SADE and GADE.

Taurins-peng, 2024-08-23:
Hi, I have a question regarding SADE and GADE. The following sentence indicates that a guest page fault occurs in this scenario.
[image]
The current case scenario is: SADE and GADE are 0, and the access type is write. The GPTE configuration is as follows:
[image]
I expect a guest page fault to occur, but the model changes the write to 0 when SADE is 0, making line 307 valid and proceeding to step_8 without reporting a guest page fault.
[image]
[image]
Could you explain this?

Taurins-peng, 2024-08-23:
[image]

ved-rivos, 2024-08-23:
When SADE is 0 then the IOMMU does not do any A/D updates in VS-stage PTEs so VS-stage PTE accesses are always reads. Since there are no stores done to VS-stage PTEs there is no need for write permission checks for the G-stage PTEs that provide address translation to those VS-stage PTEs.

Taurins-peng, 2024-08-26:
Thank you for the explanation.

closed 2024-08-26 by Taurins-peng