riscv-non-isa/riscv-iommu #408: the num of FQ  question

asdvvcc123, 2024-08-24:
Hi, 

I think  the num of FQ is wrong ,  in the refmodel you set is 9 ,  so the  num of FQ is 2^(9+1) = 1024  ;     
but in here  ,just loop 1023 time , and then said full ?    don't understand here ; 
 and why here  compare with the  FQH，  FQH here always is 0;  

[image]

thank

ved-rivos, 2024-08-24:
A queue is empty if the head is equal to the tail. A queue is full if the tail is the head minus one. The head and tail wrap around when they reach the end of the circular buffer. When queue is of size 1024 then it is full when there are 1023 records i.e. when tail is one minus head.

closed 2024-08-25 by asdvvcc123

ved-rivos, 2024-08-25:
The SMMU model implementation allows all N entries to be valid simultaneously rather than N-1 entries. However this requires tracking an additional wrap bit for the head and tail pointer each. The added complexity and state - though not high - both for SW and HW to use the one additional entry was considered not sufficiently motivated.