riscv-non-isa/riscv-iommu #411: Issues with MSI

asdvvcc123, 2024-08-25:
Hello,Sir:

I have a question regarding MSI translation. As the protocol states, the GPA needs to be translated by MSI. When there is only the S-stage (S2 is bare, S1 is not bare), I believe there should be no need for further MSI translation, right?
 
[image]

So, at this point, is it necessary to determine the translation mode? Since the protocol does not describe this here, I am not quite clear. Thank you for your reply.

ved-rivos, 2024-08-25:
> When there is only the S-stage (S2 is bare, S1 is not bare), I believe there should be no need for further MSI translation, right?

There is no option to have only S-stage. Two stage address translation is always in effect though either stage can be effectively disabled by setting its mode to Bare. Thus even when G-stage is Bare the MSI address translation if active still applies.

king729, 2024-08-26:
@ved-rivos According to the current spec, consider this scenario. Enable nest stage stage1 and use stage2 for virtual machine isolation. At this time, how does the iommu driver inside the virtual machine handle msi? Because according to the above statement, there is no situation where only stage1 is enabled. At this time, the iommu driver inside the virtual machine still needs to do MSI translation, that is, fill in the msiptp table. Is that right?

asdvvcc123, 2024-08-26:
get it, and thank a lot!

closed 2024-08-26 by asdvvcc123

ved-rivos, 2024-08-26:
@king729 

The MSI page tables are intended to provide a virtual IMSIC as defined by the advanced interrupt architecture (AIA) to a virtual machine. The virtual IMSIC is backed by either guest interrupt files in one of the harts IMSICs or backed by a memory resident interrupt file (MRIF). These virtual IMSIC interrupt files are mapped to a hypervisor selected GPAs in the guest address space. 

The MSI page tables are managed by the hypervisor and not the guest OS. Essentially the MSI page tables stand in for the G-stage page table to handle GPAs that have been classified as those of a virtual interrupt file. When in basic translate mode the functionality provided by the MSI page tables is identical to that provided by the G-stage page table but with the permissions implicitly set to R=W=U=1 and X=0. When in MRIF mode the MSI page table provides the enhanced functionality  to storing the interrupt to a memory resident interrupt file (MRIF) and generating a notice MSI. 

When the guest OS programs devices with MSI address - it is effectively programming the address - a GPA - of the seteipnum_* register of the virtual IMSIC interrupt file. The hypervisor programs the MSI page tables to map that GPA to a guest interrupt file in a physical IMSIC or to a MRIF. The guest OS itself is unaware of this remapping - it assumes it has access to an IMSIC and programs the address of the IMSIC interrupt file into devices for the purposes of generating MSIs.