riscv-non-isa/riscv-iommu #414: How to obtain interrupt ID of MSI from device

zhangdujiao, 2024-08-29:
2.3.3. presents the msi address calculation method, but I cannot see how to obtain the interrupt ID of a specific MSI from external device?
And why the 1st statement is "Let A be the GPA"? not the VA? and only DC is used, what about PC?
Many thanks!

zhangdujiao, 2024-08-30:
> The write data value and hence the interrupt ID is configured into the device as an address+data pair

That is to say, interrupt ID is the "data", and the "address" is translated to msi_address by IOMMU, is that correct?

ved-rivos, 2024-08-30:
> That is to say, interrupt ID is the "data", and the "address" is translated to msi_address by IOMMU, is that correct?

Correct.

closed 2024-08-31 by zhangdujiao

18772820305, 2025-03-12:
Hi Sir, i have few questions about MSI table:
1.If a device send a MSI that is targeted to a S-mode interrupt file, whether the MSI page table is not used during MSI address translation？
2.When a virtual hart write a MSI to another hart's VS-mode interrupt file to generate an IPI, how to determine the MSI address? Does the MMU unit of the hart need to use the MSI page table to translate GPA into PA?
[image]