riscv-non-isa/riscv-iommu #420: AD update question

asdvvcc123, 2024-09-04:
hi , sir:

[image]
here pte_changed always is 0 , so don't understand need to goto step_2 re-read pte agin? or  after change just goto step 8 , the process confuse me .

thanks

ved-rivos, 2024-09-04:
> here pte_changed always is 0 , so don't understand need to goto step_2 re-read pte agin? or after change just goto step 8 , the process confuse me .

It depends on the test.  The A/D update process is as follows:
1. Load PTE and determine if it would cause any faults.
2. If it cause any faults then  cause the fault
3. If no faults are detected then determine if A/D update is needed
4. If A/D update is needed then atomically set the A/D bit by loading the PTE from memory, setting the A or D bit and writing the PTE back to memory.

Between step 1 and 4, the PTE in memory may change changed by say the software executing on the application processor cores. If the PTE is changed then the PTE loaded in step 4 wont match the PTE loaded in step 1. This is what the code is detecting. If the PTE changed then it restarts the process from  step 1.

asdvvcc123, 2024-09-05:
thank  , get it ;

closed 2024-09-05 by asdvvcc123

yanhe234, 2024-10-31:
Hello sir, why haven't we considered Capas here What about AMO-HWAD storage? It seems that the default here is to definitely support hardware updates
```
    if ( (gpte->A == 1) && (gpte->D == 1 || is_write == 0) &&
         (gpte->D == 1 || is_implicit == 0 || gpte->W == 0 || GADE == 0 || SADE == 0) ) goto step_8;

    // A and/or D bit update needed
    if ( GADE == 0 ) return GST_PAGE_FAULT;
```

ved-rivos, 2024-10-31:
The IOMMU checks that neither GADE or SADE can be 1 if the capabilities does not support HW A/D updating. Since that has already been checked it is not required to check it again.

yanhe234, 2024-10-31:
Sorry sir, I just saw the description you just mentioned in the code.