riscv-non-isa/riscv-iommu #427: implicit access is a read or a write

Taurins-peng, 2024-09-10:
Hi Sir, how can I distinguish whether the implicit access is a read or a write? Does this relate to the read or write configuration in my case?
[image]

ved-rivos, 2024-09-10:
> how can I distinguish whether the implicit access is a read or a write?

A memory access to read or write PTEs are implicit accesses. As part of page table walk there is an implicit access to read the PTE from memory. Such reads are implicit read accesses. For a leaf PTE entry, when hardware A/D updates are supported and enabled, the IOMMU may do a write to the PTE to set the A and/or D bit. Such writes are implicit write accesses.

ved-rivos, 2024-09-11:
Hope that helped. Please ask if there are further questions.

closed 2024-09-11 by ved-rivos