riscv-non-isa/riscv-iommu #428: msi_address_translation

asdvvcc123, 2024-09-10:
hi，sir:
 
in msi_address_translation , iotval2 is never be set ; 
[image]
thanks!

ved-rivos, 2024-09-10:
The spec states:
> If the CAUSE is a guest-page fault then bits 63:2 of the zero-extended guest-physical-address are reported in iotval2[63:2]. If bit 0 of iotval2 is 1, then the guest-page-fault was caused by an implicit memory access for first-stage address translation. If bit 0 of iotval2 is 1, and the implicit access was a write then bit 1 of iotval2 is set to 1 else it is set to 0.

iotval2 is  not reported for any other cause code. 

> here i think 52 is ok 

Yes. 52 would suffice since the data and mask are pre-shifted by 12 bits but this code is not wrong as the high order 12 bits in mask will be always 0.

asdvvcc123, 2024-09-10:
thanks,get it

closed 2024-09-10 by asdvvcc123